dji-sdk/Onboard-SDK #484: M210 v2RTK OSDK 3.9 NEW_DEVICE_ERROR

snakehaihai, 2019-11-20:
Usually, For A3 NEW_DEVICE_ERROR can be solved by calling command twice and the GO app will pops up. 

However, M210 v2RTK is not working with GO app or GO 4 app. 
I tried with DJI pilot app and pilot PE4. Both cant connect. the error is

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x3040325

STATUS/1 @ getErrorCodeMessage, L752: activate
STATUS/1 @ getCMDSetActivationMSG, L801: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1424: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1431: Solutions for NEW_DEVICE_ERROR:
	* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
	* If this is a new device, you need to activate it through the App or DJI Assistant 2 with Internet
	For different aircraft, the App and the version of DJI Assistant 2 might be different
	For A3, N3, M600/Pro and M100, please use DJI GO App
	For M210 V1, please use DJI GO 4 App or DJI Pilot App
	For M210 V2, please use DJI Pilot App
	For DJI Assistant 2, it's available on the 'Download' tab of the product page
	* If this device is previously activated with another app_id and app_key, you will need to re-activate it again.
	* A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.

STATUS/1 @ getErrorCodeMessage, L752: initVehicle
STATUS/1 @ getCMDSetActivationMSG, L801: NEW_DEVICE_ERROR

STATUS/1 @ getErrorCodeMessage, L752: reset
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH
Vehicle not initialized, exiting.

dji-dev, 2019-11-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30057'>Zendesk ticket #30057</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

To activate M210 RTK V2, you need to ensure that the DJI Pilot is the latest version (and can correctly identify M210). In addition, and the need to ensure that the phone can access the network. Due to the network congestion problem existing in some local network operators, it is suggested that you can try to switch to 4G or WIFI and try again.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

snakehaihai, 2019-11-21:
Who else encounter this issue. Ignore the DJI-dev here. 

I received instruction for DJI-Qiming(cirst) from private wechat group . The activation is not working with pilot app or pilot PE app. You need to connect to both windows Dji assistant2 and ubuntu.  You have to enter assitant 2 while calling sdk.launch. 

Then it will work. (work for basic gimbal and flight control. Camera stream still not up)

closed 2019-11-21 by snakehaihai

dji-dev, 2019-11-26:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30057'>Zendesk ticket #30057</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Have you solved the problem with the Camera stream? If not, suggest you click the hyperlink on the right to fill in and submit more complete information about the problem. https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

snakehaihai, 2019-11-27:
Camera stream issue solved. Seting static IP of 42.3/24/42.2 solved the issue for me