dji-sdk/Onboard-SDK #485: L1503:Drone version not obtained while trying to run the Onboard SDK flightcontrol-sample example

Nearspace, 2019-11-20:
the issues possible reasons i have check more than ones. but the problem is going on. How can i do?
[image]

dji-dev, 2019-11-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30063'>Zendesk ticket #30063</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

In order to better understand and help you solve the problem, suggest you click the hyperlink on the right to fill in and submit more complete information about the problem.
https://formcrafts.com/a/dji-developer-feedback-en
The problem description column shall be as detailed as possible, including the following contents:
1. OSDK version and DJI aircraft model
2. OSDK development platform architecture and system version
3. Is USB-TTL used? Does it support 921600 baud? or can to try using 115200.
5. In addition, follow the five tips of the log to troubleshoot problems one by one.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

snakehaihai, 2019-11-21:
Since you are not able to provide detailed information. 

Go to  https://stackoverflow.com/questions/58688023/roslaunch-activation-failure-what-should-i-do/58760789?noredirect=1#comment104190828_58760789

Go through the checklist

closed 2021-06-24 by dji-kyle