dji-sdk/Onboard-SDK #486: dji_linux_helpers.hpp:No such file or directory

TakumiWzy, 2019-11-21:
Dear dji:
When I am using onboard-sdk-3.6.1 version, when compiling  onboard-sdk-3.6.1 / sample / Linux / advanced sensing / camera_stream_poll_sample, an error is reported: fatal error: dji_Linux_helpers.hpp: no such file or directory.   But I find the header file in sample/linux/dji_Linux_helpers.hpp,  the program can't find it ? I 'm sure the hearder path in the sample code is right.  Can you help me to fix it?

dji-dev, 2019-11-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30091'>Zendesk ticket #30091</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Currently, DJI OSDK has been updated to version 3.9. So it is recommended that you can try to use the latest version of OSDK, and can refer to the steps provided by the development document of OSDK for compilation (-DADVANCED_SENSING=ON parameter needs to be added to compile the sample related to advanced sensing). https://developer.dji.com/onboard-sdk/documentation/development-workflow/sample-setup.html#linux-onboard-computer

If the compilation results still show relevant errors, in order to better understand and help you solve the problem, need to know more information as follows:
1. OSDK version used
2. Development platform hardware architecture, system version
3. From creating the OSDK folder to performing the full log of the compilation (including all the commands you execute)

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

peza8, 2020-02-12:
Did you find a solution to this @TakumiWzy ?? I have the same issue with headers not being found.

peza8, 2020-02-13:
I figured out what was wrong in my case @TakumiWzy . I hadn't built the SDK using the `CMakeList.txt` from the root of the project. Worked after that.

closed 2021-06-24 by dji-kyle

keletso7, 2021-11-19:
@peza8 Please elaborate more on how you solved it. I have same error, headers not being found