dji-sdk/Onboard-SDK #488: TOPIC_ALTITUDE_FUSIONED 返回-155米, 怎么解决？

addriumruss, 2019-12-03:
**我订阅了话题TOPIC_ALTITUDE_FUSIONED，
结果返回的-150~-160之间，每次开机还是随机的
问题很严重，怎么解决？？？**

dji-dev, 2019-12-03:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30393'>Zendesk ticket #30393</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

为了更好的理解和帮助您解问题，需要了解更多如下信息：
1. 请问您所使用的DJI 飞行器型号是什么？以及固件版本
2. DJI OSDK 的版本
3. 如果直接运行DJI OSDK的 telemetry sample 选择“a” 选项，获得“Position (LLA) ” 数据是多少？建议提供完整log。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

addriumruss, 2019-12-03:
> Public comment from Bruce Cheung in Zendesk ticket #30393:

你好，
飞机版本： M600 Pro,
从Assistant2软件Firmware Update页面看到 固件是： V1.00.01.67
从Osdk读取的固件版本是： 3.2.41.14
OSDK: 3.9.0
 telemetry sample 选择“a” 选项情形读取的位置是： 
Position              (LLA)           = 0.559012, 2.07371, -136.644
无人机实际位置是在公司楼下的地面。

由于这个位置的完全不准确，导致飞机实际无法使用。
请尽快提供解决方案，
感谢支持！

以下是完整日志：

# ./djiosdk-telemetry-sample
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L56: Attempting to open device /dev/ttyS0 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L762: Device Serial No. = 0670155791

STATUS/1 @ parseDroneVersionInfo, L764: Hardware = PM820V3PRO

STATUS/1 @ parseDroneVersionInfo, L765: Firmware = 3.2.41.14

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x302290E

STATUS/1 @ activate, L1411: Activation successful

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ initGimbal, L926: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ initGimbal, L970: Gimbal not mounted!
| Available commands:                                            |
| [a] Get telemetry data and print                               |
| [b] Select some subscription topics to print                   |
| [c] Get telemetry data and save to file                        |
a

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=1, nTopics=1.

STATUS/1 @ startPackage, L353: Start package 1 result: 0.
STATUS/1 @ startPackage, L355: Package 1 info: freq=10, nTopics=2.

STATUS/1 @ startPackage, L353: Start package 2 result: 0.
STATUS/1 @ startPackage, L355: Package 2 info: freq=50, nTopics=2.

STATUS/1 @ startPackage, L353: Start package 3 result: 0.
STATUS/1 @ startPackage, L355: Package 3 info: freq=200, nTopics=1.

STATUS/1 @ startPackage, L353: Start package 4 result: 0.
STATUS/1 @ startPackage, L355: Package 4 info: freq=5, nTopics=5.
Counter = 0:
-------
Flight Status                         = 0
Position              (LLA)           = 0.559012, 2.07371, -136.644
RC Commands           (r/p/y/thr)     = 0, 0, 0, 0
Velocity              (vx,vy,vz)      = -0.000280894, -6.95606e-05, 0.00194299
Attitude Quaternion   (w,x,y,z)       = 0.993259, -0.00420817, 0.0082643, 0.115548
RTK if available   (lat/long/alt/velocity_x/velocity_y/velocity_z/yaw/yaw_info/pos_info) =0,0,0,0,0,0,0,
-------

Counter = 5:
-------
Flight Status                         = 0
Position              (LLA)           = 0.559012, 2.07371, -136.644
RC Commands           (r/p/y/thr)     = 0, 0, 0, 0
Velocity              (vx,vy,vz)      = -0.000280894, -6.95606e-05, 0.00194299
Attitude Quaternion   (w,x,y,z)       = 0.993259, -0.00420821, 0.00826765, 0.115549
RTK if available   (lat/long/alt/velocity_x/velocity_y/velocity_z/yaw/yaw_info/pos_info) =0,0,0,0,0,0,0,
-------

Counter = 10:
-------
Flight Status                         = 0
Position              (LLA)           = 0.559012, 2.07371, -136.644
RC Commands           (r/p/y/thr)     = 0, 0, 0, 0
Velocity              (vx,vy,vz)      = -0.000280894, -6.95606e-05, 0.00194299
Attitude Quaternion   (w,x,y,z)       = 0.993259, -0.00421029, 0.00827022, 0.115548
RTK if available   (lat/long/alt/velocity_x/velocity_y/velocity_z/yaw/yaw_info/pos_info) =0,0,0,0,0,0,0,
-------

Tao Zhijun
13905183282

 
 
  
 
DJI OSDK 的版本

addriumruss, 2019-12-03:
你好，
飞机版本： M600 Pro,
从Assistant2软件Firmware Update页面看到 固件是： V1.00.01.67
从Osdk读取的固件版本是： 3.2.41.14
OSDK: 3.9.0
 telemetry sample 选择“a” 选项情形读取的位置是：
Position              (LLA)           = 0.559012, 2.07371, -136.644
无人机实际位置是在公司楼下的地面。

由于这个位置的完全不准确，导致飞机实际无法使用。
请尽快提供解决方案，
感谢支持！

以下是完整日志：

# ./djiosdk-telemetry-sample
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L56: Attempting to open device /dev/ttyS0 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L762: Device Serial No. = 0670155791

STATUS/1 @ parseDroneVersionInfo, L764: Hardware = PM820V3PRO

STATUS/1 @ parseDroneVersionInfo, L765: Firmware = 3.2.41.14

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x302290E

STATUS/1 @ activate, L1411: Activation successful

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ initGimbal, L926: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ initGimbal, L970: Gimbal not mounted!
| Available commands:                                            |
| [a] Get telemetry data and print                               |
| [b] Select some subscription topics to print                   |
| [c] Get telemetry data and save to file                        |
a

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=1, nTopics=1.

STATUS/1 @ startPackage, L353: Start package 1 result: 0.
STATUS/1 @ startPackage, L355: Package 1 info: freq=10, nTopics=2.

STATUS/1 @ startPackage, L353: Start package 2 result: 0.
STATUS/1 @ startPackage, L355: Package 2 info: freq=50, nTopics=2.

STATUS/1 @ startPackage, L353: Start package 3 result: 0.
STATUS/1 @ startPackage, L355: Package 3 info: freq=200, nTopics=1.

STATUS/1 @ startPackage, L353: Start package 4 result: 0.
STATUS/1 @ startPackage, L355: Package 4 info: freq=5, nTopics=5.
Counter = 0:
-------
Flight Status                         = 0
Position              (LLA)           = 0.559012, 2.07371, -136.644
RC Commands           (r/p/y/thr)     = 0, 0, 0, 0
Velocity              (vx,vy,vz)      = -0.000280894, -6.95606e-05, 0.00194299
Attitude Quaternion   (w,x,y,z)       = 0.993259, -0.00420817, 0.0082643, 0.115548
RTK if available   (lat/long/alt/velocity_x/velocity_y/velocity_z/yaw/yaw_info/pos_info) =0,0,0,0,0,0,0,
-------

Counter = 5:
-------
Flight Status                         = 0
Position              (LLA)           = 0.559012, 2.07371, -136.644
RC Commands           (r/p/y/thr)     = 0, 0, 0, 0
Velocity              (vx,vy,vz)      = -0.000280894, -6.95606e-05, 0.00194299
Attitude Quaternion   (w,x,y,z)       = 0.993259, -0.00420821, 0.00826765, 0.115549
RTK if available   (lat/long/alt/velocity_x/velocity_y/velocity_z/yaw/yaw_info/pos_info) =0,0,0,0,0,0,0,
-------

Counter = 10:
-------
Flight Status                         = 0
Position              (LLA)           = 0.559012, 2.07371, -136.644
RC Commands           (r/p/y/thr)     = 0, 0, 0, 0
Velocity              (vx,vy,vz)      = -0.000280894, -6.95606e-05, 0.00194299
Attitude Quaternion   (w,x,y,z)       = 0.993259, -0.00421029, 0.00827022, 0.115548
RTK if available   (lat/long/alt/velocity_x/velocity_y/velocity_z/yaw/yaw_info/pos_info) =0,0,0,0,0,0,0,
-------

dji-dev, 2019-12-04:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30393'>Zendesk ticket #30393</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

TOPIC_ALTITUDE_FUSIONED 使用的是 ICAO model，该模型是基于WG 84坐标系，高度是以WGS 84椭球为参考面的高度，与我国海拔测量采用的不是一个同一个标准，因此会存在我国部分地区测量到的海拔是负数的情况。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

addriumruss, 2020-06-16:
不是一个标准，南京这边海拔25米左右，拿到-155就是合理的？？？那块陆地上海拔还能低于0， 什么破解释？？？

dji-kyle, 2021-06-24:
Note：
[image]
如果问题依旧存在可以再次开启

closed 2021-06-24 by dji-kyle