dji-sdk/Onboard-SDK #490: Compilation Error

DimaNyumovich, 2019-12-08:
[image]

ubuntu 16.04 
x86_64

sudo cmake .. -DADVANCED_SENSING=ON  w/o errors

sudo apt-get install nvidia-cuda-dev throw error (see pic too)

/home/dji/Desktop/Onboard-SDK3.8-20190529/onboardsdk-internal/onboardsdk/onboardsdk/closed-source/advanced-sensing/platform/src/linux_usb_device.cpp:158: undefined reference to `libusb_bulk_transfer'
collect2: error: ld returned 1 exit status
sample/platform/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/build.make:180: recipe for target 'bin/djiosdk-cameragimbal-sample' failed
make[2]: *** [bin/djiosdk-cameragimbal-sample] Error 1
CMakeFiles/Makefile2:188: recipe for target 'sample/platform/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/all' failed
make[1]: *** [sample/platform/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

dji-dev, 2019-12-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30594'>Zendesk ticket #30594</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

For the Advanced Sensing feature, you will need to add an udev file to allow your system to obtain permission and to identify DJI USB port. You can refer to https://developer.dji.com/onboard-sdk/documentation/development-workflow/environment-setup.html#ubuntu-linux
And need your system to install libusb, can refer to https://developer.dji.com/onboard-sdk/documentation/M210-Docs/oes-checklist.html#software-dependencies-for-advanced-sensing

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

DimaNyumovich, 2019-12-12:
thx it helped

closed 2021-06-24 by dji-kyle