dji-sdk/Onboard-SDK #491: i want migration dji onboard sdk 是她们operating System

programdog, 2019-12-17:


programdog, 2019-12-17:
i want migration dji onboard sdk stm32 migration to RTOS ， so Is it feasible？ and if you have RTOS 
 solutions， please tell me thx!
我想把dji onboard sdk stm32 加入RTOS ， 所以想问一下可行性， 或者是否有现有的，与RTOS结合的方案？

dji-dev, 2019-12-18:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30860'>Zendesk ticket #30860</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

很抱歉，目前DJI Onboard SDK 提供的STM32 Sample为裸机程序，暂没有提供RTOS的样例，建议您可以根据实际项目需求进行代码移植。
或者，如果您有相关需求，建议您点击右侧的超链接填写相关需求信息， https://formcrafts.com/a/dji-developer-feedback-cn
为了更好的匹配您的需求，填写时尽量包括如下内容：
1. 期望的功能和需求
2. 公司/机构简介
3. 使用此特性需求的项目描述
4. 个人背景或简介

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

programdog, 2019-12-18:
> Public comment from Bruce Cheung in [Zendesk ticket #30860](https://djisdksupport.zendesk.com/agent/tickets/30860):
> 
> 尊敬的客户，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 很抱歉，目前DJI Onboard SDK 提供的STM32 Sample为裸机程序，暂没有提供RTOS的样例，建议您可以根据实际项目需求进行代码移植。
> 或者，如果您有相关需求，建议您点击右侧的超链接填写相关需求信息， https://formcrafts.com/a/dji-developer-feedback-cn
> 为了更好的匹配您的需求，填写时尽量包括如下内容：
> 
> 1. 期望的功能和需求
> 2. 公司/机构简介
> 3. 使用此特性需求的项目描述
> 4. 个人背景或简介
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持

ok thx , i will to try and research

closed 2021-06-24 by dji-kyle