dji-sdk/Onboard-SDK #492: fix: remove the api about avoid obstacle

KyubeomLee, 2019-12-19:
fix: remove the api about avoid obstacle//commit 5e38044cf41583acf451da909d9829b86d0988b2

in api document describe use collision avoidance,
but in code why remove that api?

dji-dev, 2019-12-19:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30923'>Zendesk ticket #30923</a>:<br><br>Dear customer,

Thank you for contacting DJI.

Thank you for your feedback on this problem. We will further confirm the problem and inform you as soon as any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2019-12-24:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30923'>Zendesk ticket #30923</a>:<br><br>Dear customer,

Thank you for contacting DJI.

 Sorry that this feature is removed because it is not compatible with the current firmware, but will be back online in the next version of OSDK.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle