dji-sdk/Onboard-SDK #493: setWaypointEventCallback为什么不执行，如何配置？

maxiao99, 2019-12-26:
在OSDK的mission_sample例程中，用setWaypointEventCallback来注册回调函数，为什么回调函数不执行，是否还需要其他配置？非常感谢
[image]

dji-dev, 2019-12-26:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31104'>Zendesk ticket #31104</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

根据您提供的信息暂无法判断问题原因，为了更好的了解和帮助您解决问题，建议您点击右侧的超链接填写更详细的问题信息（包括完整的程序执行log）， https://formcrafts.com/a/dji-developer-feedback-cn

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

maxiao99, 2019-12-27:
飞行器对应M600 Pro，在Onboard-SDK/sample/platform/linux/missions目录下只添加回调函数，如下图所示，其它代码未作修改，为什么setWaypointEventCallback注册的回调函数不执行？
[image]

maxiao99, 2019-12-27:
飞行器对应M600 Pro，在Onboard-SDK/sample/platform/linux/missions目录下main.cpp文件只添加回调函数，如下图所示，其它代码未作修改，为什么setWaypointEventCallback注册的回调函数不执行？
[image]

dji-dev, 2019-12-27:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31104'>Zendesk ticket #31104</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

经过实际测试和验证无法复现您所描述的问题现象，测试环境/步骤/代码可以参考如下：
飞行平台：M600 Pro，固件版本 3.2.41.14
开发平台：DJI Manifold2-G Ubuntu16.04
OSDK版本：DJI Onboard SDK 3.9
测试步骤：
1. 连接M600 Pro到PC，并运行DJI Assistant2 模拟器
2. 修改DJI Onboard SDK 3.9  missions_sample main.cpp文件（完整的测试代码附文后）
3. 重新编译并执行 missions sample
4. 可以观察到OSDK程序输出初始化mission的信息，模拟器中M600开始起飞并执行航点任务，OSDK程序进入回调并打印信息
建议您可以直接使用文后提供的测试代码，并参考如上测试步骤重新验证。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

测试代码（Onboard SDK 3.9  missions_sample main.cpp）：

    #include "mission_sample.hpp"

    using namespace DJI::OSDK;
    using namespace DJI::OSDK::Telemetry;

    /*[DJI Bruce] --------------- Code Add Start ------------------ [2019.0919]*/
    void missionEventCallback(Vehicle* vehicle,RecvContainer recvFrame,UserData userData);
    /*[DJI Bruce] --------------- Code Add End ------------------ [2019.0919]*/

    int
    main(int argc, char** argv)
    {
      // Initialize variables
      int functionTimeout = 1;

      // Setup OSDK.
      LinuxSetup linuxEnvironment(argc, argv);
      Vehicle*  vehicle = linuxEnvironment.getVehicle();
      if (vehicle == NULL)
      {
        std::cout << "Vehicle not initialized, exiting.\n";
        return -1;
      }

      // Obtain Control Authority
      vehicle->obtainCtrlAuthority(functionTimeout);

      // Setup variables for use
      uint8_t wayptPolygonSides;
      int    hotptInitRadius;
      int    responseTimeout = 1;

      // Display interactive prompt
      std::cout
        << "| Available commands:                                            |"
        << std::endl;
      std::cout
        << "| [a] Waypoint Mission                                          |"
        << std::endl;
      std::cout
        << "| [b] Hotpoint Mission                                          |"
        << std::endl;
      char inputChar;
      std::cin >> inputChar;
      switch (inputChar)
      {
        case 'a':
          // Waypoint call
          wayptPolygonSides = 6;
          runWaypointMission(vehicle, wayptPolygonSides, responseTimeout);
          /*[DJI Bruce] --------------- Code Add Start ------------------ [2019.0919]*/
          std::cout << "DJI Bruce: setWaypointEventCallback\n";
          vehicle->missionManager->wpMission->setWaypointEventCallback(missionEventCallback,0);
          while(1){}
          /*[DJI Bruce] --------------- Code Add End ------------------ [2019.0919]*/
          break;
        case 'b':
          hotptInitRadius = 10;
          runHotpointMission(vehicle, hotptInitRadius, responseTimeout);
          break;
        default:
          break;
      }

      return 0;
    }

    /*[DJI Bruce] --------------- Code Add Start ------------------ [2019.0919]*/
    void missionEventCallback(Vehicle* vehicle,RecvContainer recvFrame, UserData userData){
      DSTATUS("WP Event Callback, \tincident_type:%d \twaypoint_index:%d \tcurrent_status:%d \n",
        recvFrame.recvData.wayPointReachedData.incident_type,
        recvFrame.recvData.wayPointReachedData.waypoint_index,
        recvFrame.recvData.wayPointReachedData.current_status);
    }
    /*[DJI Bruce] --------------- Code Add End ------------------ [2019.0919]*/

maxiao99, 2019-12-30:
您好，按照您的操作重新git下载执行，但回调函数还是不会执行，请问该功能是否对普通用户开放，在源代码cmake时是否需要其它操作？
[image]
[image]

dji-kyle, 2021-06-24:
若回调不执行，可以检查并勾选DJI Assistant2上的地面状态推送再测试。
[image]

closed 2021-06-24 by dji-kyle