dji-sdk/Onboard-SDK #496: Meaning of values in TOPIC_GPS_SIGNAL_LEVEL topic

L4ncelot, 2020-01-15:
Hi,

I'm using the latest version of **DJI OSDK** and I would like to ask a question for clarification. 

As described in documentation [here](https://developer.dji.com/onboard-api-reference/group__telem.html), `TOPIC_GPS_SIGNAL_LEVEL` topic returns values from 0 to 5 based on **GPS** health and must be greater than 3 for a 3D fix. I couldn't find anywhere precisely what those numbers mean (e.g. 3 = 3D fix, 5 = RTK 3D fix, etc.).

Could you please provide me with this information? It would be really useful. 

Thank you in advance.

dji-dev, 2020-01-16:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31638'>Zendesk ticket #31638</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry, we did n’t understand what you meant. Where did you see “3 = 3D fix, 5 = RTK 3D fix”? TOPIC_GPS_SIGNAL_LEVEL is only used to describe the quality of GPS signals.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

L4ncelot, 2020-01-16:
Sorry for the confusion. I don't see those values anywhere. It was just an example of what the values could probably mean. So let me clarify this. Is there any way to determine **GPS** fix type (3D fix, RTK fix, etc.) based on the information provided by the quality of the **GPS** signal or other topic?

dji-dev, 2020-01-19:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31638'>Zendesk ticket #31638</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry to keep you waiting, the status of GPS and RTK is separate. For the fix status of GPS, you can use TOPIC_GPS_DETAILS. For the fix status of RTK, you can use TOPIC_RTK_CONNECT_STATUS.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

L4ncelot, 2020-01-20:
Ah... now I see it. I must have overlooked the **GPSDetail** struct. Thank you.

closed 2020-01-20 by L4ncelot