dji-sdk/Onboard-SDK #498: Get waypoint info...

DimaNyumovich, 2020-01-20:
Hi all!
I'd like to ask about one problem that I can't find decision... I have situation when I made upload waypoints mission and made start. I want to know if I have possibility to get waypoints info after my application failed and I restarted it, but without uploading waypoints again.

Best regards,
Dmitry.

dji-dev, 2020-01-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31754'>Zendesk ticket #31754</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

You can call setWaypointEventCallback to register the callback function to determine if the aircraft has reached the waypoint. It can also be used to determine whether the upload and execution are successful. The code sample is as below.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/JM2dtGVmqNYq2K3CXOnAAQJ7p' target='zd_attach'>inline79152903.png</a>

DimaNyumovich, 2020-01-21:
I did it but got Signal: SIGSEGV (Segmentation fault)
(
my configuration 
ubuntu 16 
intel CPU
current sdk from git
)
my code :
....
Response_complex_additional_waypoint_info wayPoint_info(Vehicle *vehicle, int responseTimeout) {
    Response_complex_additional_waypoint_info res;
    RecvContainer recvFrame;
    UserData userData;
    vehicle->missionManager->wpMission->setWaypointEventCallback(missioneventcallback, 0);
    if (vehicle->missionManager->wpMission != NULL) {
        std::cout << "Waypoint Mission info.\n";
        std::vector<WaypointMission *> &ggg = vehicle->missionManager->wpMissionVector;

        unsigned long missionSize = ggg.size();
        if (missionSize != 0) {
            uint8_t pointsNumber = ggg[ggg.size() - 1]->getWaypointSettings(10).data.indexNumber;
            for (int i = 0; i < pointsNumber; i++) {
                const ACK::WayPointIndex &pointData = vehicle->missionManager->wpMission->getIndex(i, 10);
                res.additionalData.push_back(pointData);
            }
            res.success = true;
            
        } else {
            res.message.push_back("Didn't upload any mission");
        }
    } else {
        res.message.push_back("Didn't upload any mission");
    }
    return res;
}
....
// callback function
void missioneventcallback(Vehicle* vehicle, RecvContainer recvFrame, UserData userData){
    DSTATUS("wp event callback, \tincident_type:%d\twaypoint_index:%d\tcurrent_status:%d\n",
            recvFrame.recvData.wayPointReachedData.incident_type,
            recvFrame.recvData.wayPointReachedData.waypoint_index,
            recvFrame.recvData.wayPointReachedData.current_status);
}

dji-dev, 2020-01-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31754'>Zendesk ticket #31754</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

​It is recommended that you can directly use the following code for testing and verification, replace the content of OSDK-3.9/sample/platform/linux/missions/main.cpp, and then recompile.

    #include "mission_sample.hpp"

    using namespace DJI::OSDK;
    using namespace DJI::OSDK::Telemetry;

    /*[DJI Bruce] --------------- Code Add Start ------------------ [2019.0919]*/
    void missionEventCallback(Vehicle* vehicle,RecvContainer recvFrame,UserData userData);
    /*[DJI Bruce] --------------- Code Add End ------------------ [2019.0919]*/

    int
    main(int argc, char** argv)
    {
      // Initialize variables
      int functionTimeout = 1;

      // Setup OSDK.
      LinuxSetup linuxEnvironment(argc, argv);
      Vehicle*  vehicle = linuxEnvironment.getVehicle();
      if (vehicle == NULL)
      {
        std::cout << "Vehicle not initialized, exiting.\n";
        return -1;
      }

      // Obtain Control Authority
      vehicle->obtainCtrlAuthority(functionTimeout);

      // Setup variables for use
      uint8_t wayptPolygonSides;
      int    hotptInitRadius;
      int    responseTimeout = 1;

      // Display interactive prompt
      std::cout
        << "| Available commands:                                            |"
        << std::endl;
      std::cout
        << "| [a] Waypoint Mission                                          |"
        << std::endl;
      std::cout
        << "| [b] Hotpoint Mission                                          |"
        << std::endl;
      char inputChar;
      std::cin >> inputChar;
      switch (inputChar)
      {
        case 'a':
          // Waypoint call
          wayptPolygonSides = 6;
          runWaypointMission(vehicle, wayptPolygonSides, responseTimeout);
          /*[DJI Bruce] --------------- Code Add Start ------------------ [2019.0919]*/
          std::cout << "DJI Bruce: setWaypointEventCallback\n";
          vehicle->missionManager->wpMission->setWaypointEventCallback(missionEventCallback,0);
          while(1){}
          /*[DJI Bruce] --------------- Code Add End ------------------ [2019.0919]*/
          break;
        case 'b':
          hotptInitRadius = 10;
          runHotpointMission(vehicle, hotptInitRadius, responseTimeout);
          break;
        default:
          break;
      }

      return 0;
    }

    /*[DJI Bruce] --------------- Code Add Start ------------------ [2019.0919]*/
    void missionEventCallback(Vehicle* vehicle,RecvContainer recvFrame, UserData userData){
      DSTATUS("WP Event Callback, \tincident_type:%d \twaypoint_index:%d \tcurrent_status:%d \n",
        recvFrame.recvData.wayPointReachedData.incident_type,
        recvFrame.recvData.wayPointReachedData.waypoint_index,
        recvFrame.recvData.wayPointReachedData.current_status);
    }
    /*[DJI Bruce] --------------- Code Add End ------------------ [2019.0919]*/

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

DimaNyumovich, 2020-01-27:
Hello again.
I try again to describe my problem. Imagine the situation that my application fell during the mission. After restoring the application, I need to get information from the drone - in what phase of the mission it is. How can i do this?

 In addition, mission information has already been loaded into the drone controller. In this case, after the application is restored, I have to do uploading waypoints again. This means that the mission should begin from the beginning. I understand it that way.

So there is a final question. Is there a mechanism to restore drone control after losing contact with it without restarting the mission with the possibility of obtaining the current state of the mission?

Best regards,
Dmitry

dji-dev, 2020-01-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31754'>Zendesk ticket #31754</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the problem information you provided, and will analyze and deal with it after the Chinese Spring Festival (1-23 ~ 2-9).

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-02-14:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31754'>Zendesk ticket #31754</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

1. Imagine the situation that my application fell during the mission. After restoring the application, I need to get information from the drone - in what phase of the mission it is. How can i do this?
==> If the waypoint task created by the DJI Onboard SDK, you can also register waypoint callbacks to get the current task status in real-time, and in the event of task failure, you can recreate and upload new tasks from the point of failure. (this is the advice provided to you, which can be realized through SDK development)

2. In addition, mission information has already been loaded into the drone controller. In this case, after the application is restored, I have to do uploading waypoints again. This means that the mission should begin from the beginning. I understand it that way.
==> Your understanding is correct, but you can refer to the answer suggestion in question 1.

3. So there is a final question. Is there a mechanism to restore drone control after losing contact with it without restarting the mission with the possibility of obtaining the current state of the mission?
==> If the waypoint task created using the DJI Onboard SDK and start the SDK out of control strategy, the waypoint task execution will not be affected when the remote control is out of control/recovers, and OSDK will continue to obtain the task status.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle