dji-sdk/Onboard-SDK #499: Wrong condition for landing

hdino, 2020-01-23:
The condition to check if the landing was successful makes no sense because it is always true.
https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/sample/platform/linux/flight-control/flight_control_sample.cpp#L695-L698

dji-dev, 2020-01-24:
Public comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/31803'>Zendesk ticket #31803</a>:<br><br>Our engineers will be on holiday for the Chinese Spring Festival (1-23 ~ 1-30), which may cause some delays in responding to your issue. We appreciate your patience and apologize for the inconvenience this may cause you.

dji-dev, 2020-01-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31803'>Zendesk ticket #31803</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the problem information you provided, and will analyze and deal with it after the Chinese Spring Festival (1-23 ~ 2-9).

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-02-14:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31803'>Zendesk ticket #31803</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have upgraded your problem to the corresponding R&D engineer. Because of the novel coronavirus prevention and traffic controls in China, most of our SDK engineers are telecommuting. So the resolution of your problem will be delayed. We appreciate your patience and apologize for the inconvenience.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-04-08:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31803'>Zendesk ticket #31803</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The logic of the aircraft is that it will automatically cut to the GPS mode after a successful landing, and this test is done to check whether the flight control logic is normal to prevent the motor from rotating.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle