dji-sdk/Onboard-SDK #501: telemetry for gimbal

DimaNyumovich, 2020-01-29:
One more my problem...

When I try to use your sample for gimbal I see that camera moves. At the end of operations it has position not "0" for yaw, pitch and roll. But when I start my application with the same position of camera I get always telemetry with yaw, pitch and roll = 0. What can be wrong?

Best regards,
Dmirty.

dji-dev, 2020-02-03:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31858'>Zendesk ticket #31858</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

According to the information you provided, we are unable to determine the cause of the problem. In order to better understand and help you solve the problem, it is recommended that you click the hyperlink on the right to fill in more detailed problem information (including the necessary log/screenshot/video), https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

MarkinYuriy, 2020-02-10:
Hi
have I a method from any class to get the info that Gimbal is not found?
I  see it log, but how can I  use it in my app?

closed 2021-06-24 by dji-kyle