dji-sdk/Onboard-SDK #504: JoyStick

MarkinYuriy, 2020-02-04:
Hi
I have 600 Pro
Ubuntu 16
OSDK 3.9

I have a problem with orders as a joystick
I mean I want to send axes x,y,z and yaw
What the method and in what the class  I can use for that?
Can some send working examples?

dji-dev, 2020-02-05:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31947'>Zendesk ticket #31947</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It is recommended that you use the "flightCtrl()" API provided by DJI OSDK. For the usage and description of the API, please refer to   https://developer.dji.com/onboard-api-reference/classDJI_1_1OSDK_1_1Control.html#a1f2c5a277f091fb5fb5b39baf6f6fa0f

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

MarkinYuriy, 2020-02-10:
Thank you for answer
I need to use analog from DJI_SDK ROS
http://wiki.ros.org/dji_sdk(
/dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition (sensor_msgs/Joy)
Command the roll pitch angle, height, and yaw rate.)
Can you send an example that implements it?

in ROS I used [x,y,z,yaw] axes for example [0.1,0,15,0]; x,y,yaw-[-1:1], z-altitude in meters for   
publish it and saw that drone flight as I asked

I mean what needs to send in words of CtrlData (flag, x, y, z, yaw)
First of all I - what flag I need to use?
And what is x, y, z, yaw relevant for my example above?
Additionally, if I have trouble with GPS can I use CtrlData?

dji-dev, 2020-02-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31947'>Zendesk ticket #31947</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It is recommended that you can refer to the ROS command below

    rostopic pub -r 50 /dji_sdk/flight_control_setpoint_ENUposition_yaw sensor_msgs/Joy "{header:{seq: 0,stamp:{secs: 0,nsecs: 0},frame_id: ''},axes:[1,1,1,1],buttons:[0]}"

About the meaning of "CtrlData", it is recommended that you can refer to https://developer.dji.com/onboard-api-reference/classDJI_1_1OSDK_1_1Control.html#a1f2c5a277f091fb5fb5b39baf6f6fa0f
If in the absence of GPS, horizontal control can be controlled by using Angle or angular rate.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

MarkinYuriy, 2020-02-23:
Thank you for answer but it's difficult to understand 
Can you simple show example to use ctrlData method  for my case
I need to send x,y,z,yaw exactly as from stick 

if use ROS it will be simple rostopic pub -r 50 /dji_sdk/flight_control_setpoint_ENUposition_yaw sensor_msgs/Joy "{header:{seq: 0,stamp:{secs: 0,nsecs: 0},frame_id: ''},axes:[x,y,z,yaw],buttons:[0]}"

how it will look in case ctrlData?
in manual it's not clear... sorry

closed 2021-06-24 by dji-kyle

aurtodome, 2022-07-14:
so how ah