dji-sdk/Onboard-SDK #505: Clarification on flightCtrl API input

kanglei1130, 2020-02-20:
enum HorizontalLogic
  {
    /*!
     - Set the control-mode to control pitch & roll
     angle of the vehicle.
     - Need to be referenced to either the ground or
     body frame by HorizontalCoordinate setting.
     - Limit: 35 degree
     */
    HORIZONTAL_ANGLE = 0x00,
    /*!
     - Set the control-mode to control horizontal
     vehicle velocities.
     - Need to be referenced to either the ground
     or body frame by HorizontalCoordinate setting.
     - Limit: 30 m/s
     */
    HORIZONTAL_VELOCITY = 0x40,
  };

To control the pitch and roll, we have to use the flightCtrl API with HorizontalLogic. 

If I use HORIZONTAL_ANGLE, and set the input value to be 1.0 on pitch, it will pitch 1.0 degree, with some errors of course, my question is, does the error to be somehow consistent? Say if I input 1.0, it results in 1.1 degree, and it will be 1.1 degree all the time (with some errors of course)? If we do not consider about wind.

Similarly, if I HORIZONTAL_VELOCITY, and set the input value to be 1.0 on x, does it always fly at 1m/s with consistent errors? Or it is actually affected by the accuracy of GPS? If the GPS jumped 1meter, hypothetically, my wanted speed is 1m/s, because the GPS jumped, the actual speed become 2m/s or 0m/s?

dji-dev, 2020-02-20:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32256'>Zendesk ticket #32256</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have upgraded your problem to the corresponding R&D engineer. Because of the novel coronavirus prevention and traffic controls in China, most of our SDK engineers are telecommuting. So the resolution of your problem will be delayed. We appreciate your patience and apologize for the inconvenience.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2021-07-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/32256'>Zendesk ticket #32256</a>:<br><br>Dear developer,

Thank you for contacting DJI.

Sorry for the trouble caused to you. Recently we are dealing with remaining issues. We noticed that you have an issue and want to confirm with you whether the issue still exists:
Clarification on flightCtrl API input

We confirm that the information is synchronized with you as follows：
The errors of attitude and speed are not fixed, and are affected by many factors, such as wind speed, GPS, etc. The attitude is mainly affected by wind speed. Speed is affected by IMU, GPS, vision, etc.

The error in the simulator is constant. The simulator can only be used to simulate some basic functions. Users of accuracy-related data need to obtain actual test data, which cannot be based on the simulator.

The error of the attitude command is within 1deg if there is no wind, it can reach 2~4deg if there is wind, and the speed is about 0.3m/s. If the GPS signal is not good, but the vision works, the accuracy is guaranteed. if both GPS and vision are abnormal, then the attitude cannot be controlled at this time.

If the problem has been resolved, please ignore this email.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

closed 2021-07-06 by dji-kyle