dji-sdk/Onboard-SDK #506: Control latency of the flightCtrl API

kanglei1130, 2020-02-20:
When I use the flightCtl API, it seems there is 1 second latency. Say if I request the drone to decend with 1m/s, the time between I send out the command and the time of the z-speed become 1m/s, is about 1 second, which is pretty huge for my application.

May I ask if this is by design and any way I can optimize for this please? 

What is the expected latency should I see please?

kanglei1130, 2020-02-20:
I am using DJI M100, with latest firmware.

There is also a Virtual RC functionality, does it have lower delay?

My question is, what is the lowest control latency I can get and what is the best API for it, please?

dji-dev, 2020-02-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32281'>Zendesk ticket #32281</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that the Virtual RC feature of DJI M100 is no longer supported and is not recommended.
It is recommended that you use The flightCtrl API interface provided by DJI OSDK. For the problem of the lowest control latency, it is suggested to refer to the following maximum support frequencies for the control functions:

    Control::positionAndYawCtrl # The highest frequency 50Hz
    Control::velocityAndYawRateCtrl # The highest frequency 50Hz
    Control::attitudeAndVertPosCtrl # The highest frequency 200Hz But the vertical position information is only updated at 50Hz
    Control::angularRateAndVertPosCtrl # The top cycle frequency is 400Hz but the vertical position information is only updated at 50Hz
    Control::flightCtrl # According to the incoming "flag", the highest update frequency of angularRate information is 400Hz, attitude information is 200Hz, speed and position information is 50Hz.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

kanglei1130, 2020-02-21:
Thanks for your response.
Just to be clear, if I can flight ctrl to request it fly at 0.1 m/s, and I send the API at time 10ms, when the drone receives the command and fly at 0.1m/s, 20ms or 0.5s or something else? I meant the latency between I call the API and the time the drone made a move.

dji-dev, 2020-02-24:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32281'>Zendesk ticket #32281</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that the real response time of different models may be different, so we can't provide you with accurate response time data. However, the fastest response frequency is 50HZ (20ms) for horizontal speed control using the "flight Ctrl" API, so the speed control of 0.1m/s is possible, but the response time of 10ms is not supported. And it is recommended that you can use this API for actual validation and testing.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle