dji-sdk/Onboard-SDK #509: Flight control rates clarification

L4ncelot, 2020-03-11:
Hi,

I'm using A3 autopilot and DJI OSDK API to control the UAV with my application. I need to control the vehicle with a frequency of 100 Hz.

From issues mentioned before the frequencies are defined as:
```
Control::positionAndYawCtrl # The highest frequency 50Hz
Control::velocityAndYawRateCtrl # The highest frequency 50Hz
Control::attitudeAndVertPosCtrl # The highest frequency 200Hz But the vertical position information is only updated at 50Hz
Control::angularRateAndVertPosCtrl # The top cycle frequency is 400Hz but the vertical position information is only updated at 50Hz
Control::flightCtrl # According to the incoming "flag", the highest update frequency of angularRate information is 400Hz, attitude information is 200Hz, speed and position information is 50Hz.
```
This is a control flag I set:
```
uint8_t flag = (Control::HORIZONTAL_ANGULAR_RATE | Control::VERTICAL_THRUST | Control::YAW_RATE |
                    Control::HORIZONTAL_BODY | Control::STABLE_DISABLE);
```

I'm calling the control function `flightCtrl` with `roll_rate`, `pitch_rate`, `yaw_rate` and `thrust` with the frequency of 100 Hz. From the information provided above, this should theoretically run at 400 Hz max, but I can see some kind of delay between calling the function and reaction of the UAV.

Can you please clarify what the sentence `"The top cycle frequency is 400Hz but the vertical position information is only updated at 50Hz"` means? Does it mean every control loop has 50 Hz limitations because autopilot doesn't update the vertical part on higher frequencies? What does the autopilot do with messages on higher frequencies than the maximum update rate? Does it ignore them? Does it ignore only the vertical part, but executes given attitude rates? 

Thank you for response in advance.

dji-dev, 2020-03-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The question needs further analysis and confirms, if need further communication or there is any result, we will contact you immediately.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

L4ncelot, 2020-03-13:
Here are some graphs to demonstrate the problem I have with the controller. I have my angular rates restricted to 150 deg/s and thrust is within range <0;1> (scaled for DJI with a factor of 100).

Here you can see body rates being pretty much 0, because I'm flying only in the in Z axis.

[image]
[image]
[image]

Graphs below demonstrate the problem I have. I send a navigation goal to my application which then starts to increase the thrust (time 1s). To problem is that the autopilot doesn't react and locks itself for a period of time until it gives back the control (12s).

I thought that maybe the flight controller doesn't like the rate of change in the thrust seen in the beginning. But my other runs show that it will lock itself even with a slow increase of thrust without the bump seen in 1s. Also, I thought there's maybe some kind of safety check within 1m above the ground. That's not the case, because the problem occurs even in the altitude of 50m.

[image]
[image]
[image]

Is it some kind of safety feature of the flight controller? Or could be the problem somewhere else? We're connected to the autopilot through FTDI. So I've tried to measure the bandwidth of the serial link and it's pretty much a few kbit/s. We're using a baud rate of 921600, so it shouldn't be a problem. 

It is very crucial for my application to be able to control the vehicle without it taking over because it messes everything up pretty much.

dji-dev, 2020-03-17:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thank you for your detailed data information. The question needs further analysis and confirms, if there is any result, we will contact you immediately.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-03-31:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

1 after confirmation, the control frequency of angular rate and the throttle is 400HZ, the control frequency of Angle is 200HZ, and the control frequency of speed and position is 50HZ.
2. From the analysis of the data you sent the second time, there may be control failures. It is suggested that you reproduce the problem and export the flight control data (exported from DJI Assitant2), so as to better help you analyze and solve the problem.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

L4ncelot, 2020-04-03:
Hi, 

I've done the testing again. 

**This graph represents thrust we're sending to the autopilot:**
[image]

**This graph represents MotorCtrl:PWM:Front from autopilot's log file:**
[image]

The autopilot log can be downloaded [here](https://drive.agentfly.com/d/s/546366787843832847/sNL-QKERPtYav2AUriIZVh6nmk3nVTKV-.73A5.cVlQc_).

Thank you for the response.

Michal Vatecha

dji-dev, 2020-04-03:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the information you provided. We will make further analysis and confirm the problems and will inform you as soon as there is any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-06-08:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

By analyzing the flight control record, it is determined that the data you recorded may be wrong. The time of the throttle and the height record provided is not aligned, and the throttle record is not completely consistent with the flight control record.
From the perspective of flight control, there was nothing unusual in the whole process. The throttle was idle at 15%, and after increasing the throttle (3-4S), take-off and hover.
As shown in the figure below, the time of flight control is not aligned with the time in the picture you provided. After increasing the throttle, the altitude remains unchanged because the aircraft is in idle speed.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/14XouZqZ7xh8e8IX4MPRHoaZD' target='zd_attach'>inline2046711718.png</a>

L4ncelot, 2020-06-08:
Hi,

thank you for looking at our issue. You're right about the time alignment. This is caused by the difference in time between `ROS` and autopilot. For an aligned demonstration of the problem have a look at the first 6 graphs I've posted (all of them are in the system time). But this doesn't really solve our problem. Let me demonstrate the issue again by another image.

The graph should be aligned as seen in the picture below. It's not accurate but it demonstrates the issue we have. When we start to increase the throttle and send it to the autopilot with the `DJI OSDK`, there no change in the throttle on the autopilot's side. Our application reads there's no change in the throttle so it's trying to increase it even more (second big throttle increase). It takes the autopilot some time to realize there is a change in the throttle. 

Could it be perhaps that the `DJI OSDK` holds the control data for some time before sending it to the autopilot for some reason causing this control lag? It would explain why there's no change in the throttle on the autopilot's side.

[image]

It's really interesting this issue doesn't appear all the time. But when it does, it really messes things up and the drone is uncontrollable and we have no idea why is this problem occurring.

Thank you again for taking the time to solve this issue.

Best regards, Michal Vatecha.

dji-dev, 2020-06-09:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the problem information, need to further analyze and confirm this phenomenon.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

L4ncelot, 2020-06-26:
Hi,

we've further analyzed the issue we currently have with a new set of data being time aligned for better understanding.

Firstly we tried to figure out whether was the `FTDI` defected or not by introducing another `FTDI` sniffer to the system. The diagram of this setup can be seen in the picture below. 

[image]

Then we parsed the control command messages coming from the `FTDI` sniffer and compared them to the control messages before they're being sent to the main `FTDI`. This would tell us whether some control messages were lost or not. As it turns out this is not the case and both `FTDIs` (main and sniffer) are working just fine as you can see in the 4 images below. Plots `osdk_*` represent the control command before being sent to the main `FTDI` via `ROS OSDK`. Plots `sniffer_*` represent control commands parsed from the sniffer FTDI.

[image]
[image]
[image]
[image]

The fact that `FTDIs` are working well was also confirmed later by analyzing the autopilot's log file itself. For this purpose, we compared the telemetry we're receiving from the autopilot, which feeds our attitude rate controller, with the control commands our controller demands from the autopilot. On top of that, we also analyzed the SDK control commands from autopilot's log file itself (columns `serial_api_inputs:sdk_roll_x`, `serial_api_inputs:sdk_pitch_y`, `serial_api_inputs:sdk_yaw`, `serial_api_inputs:sdk_thr_z`) to prove the control commands are parsed successfully by the autopilot. The link to the log file can be found [here](https://drive.agentfly.com/d/f/561564355703913431).

This is being demonstrated in the 4 pictures below. Plots `telemetry_*` represent telemetry being sent by autopilot and received in the `ROS OSDK`. Plots `setpoint_*` represent the control commands `ROS OSDK` sends to the autopilot. Plots `sdk_*` represent parsed control commands from the autopilot's log file.

[image]
[image]
[image]
[image]

To describe this test briefly we're. We're sending random waypoint commands with fixed altitude to our controller producing the control commands being sent by the `ROS OSDK`. The first result of this test confirms that the autopilot is parsing our control commands correctly because the `setpoint_*` and `sdk_*` plots are identical. Also, the telemetry most of the time follows both `setpoint_*` and `sdk_*` graphs as you can see from the pictures. This results in smooth flying without any issues whatsoever. Unfortunately, sometimes the telemetry doesn't really follow our demanded commands resulting in oscillations in the flying (Oscillations mark - copter swings back and forth during the movement as seen at the beginning of the flight). Sometimes the autopilot can recover from this incident and fly smoothly again as seen in the graphs. But other times it just ignores the control commands (Control lost mark). Our controller is confused at this point and produces big spikes in the thrust command but the autopilot ignores it and just sits still in the air.

I would assume the copter wouldn't even fly if there was some issue with tuning our controller or the `A3's` controller. This is very strange as when it's working the trajectory following is smooth. But when the error occurs it messes everything up. I hope these plots will help you to debug the issue or at least provide us with some information on what to do next. I have a few scripts prepared for analyzing autopilot's log file so don't hesitate to ask for other plots which could help you in some way.

Thank you again for taking the time to solve this issue.

Best regards, Michal Vatecha.

dji-dev, 2020-06-28:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the new experimental data and flight log data you provided, and we have submitted them to the corresponding R&D engineers for analysis. We will inform you as soon as we have the results. Thank you for your patience.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-08-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/32734'>Zendesk ticket #32734</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

About your questions ,  in order to better analyze it , there are some info needs comfirm with you.
1、about the "thrust", could you please describe how to generate it?
2、About the FTDI, could you please give more info about it or about the tool links?
3、about the following API，could you please provide the real args you used ? if it is convient, show the code is better.
 [image]​
4、we have tested by angular rate and the throttle control , but faied to reproduce this issue. If it is convient, could you please test again with the same enviroment and provides the whole drone log to the following links,
https://pan-sec.djicorp.com/apps/files/?dir=/&fileid=1113732#pdfviewer

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/D8EVCVKtKdUihq8YzMZHUMMQ0' target='zd_attach'>inline-517397977.png</a>

dji-kyle, 2020-08-27:
@L4ncelot 
Dear developer ,

About the quetion,  Flight control rates clarification ticket#32734. with last replied, i want to comfirm with you if you have received it? 

And for this question, we need your cooperation to analyze it . If it is convenient, could you please provide the log and the relevant information mentioned in the previous reply again ?

and you can upload to the DJI disk: 

https://pan-sec.djicorp.com/s/M4rYo76GcPQkZi2
password: dji123

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

L4ncelot, 2020-08-28:
Hi @dji-kyle, I just wanted to confirm we received the last reply. I've just uploaded the autopilot log data to your disk as you asked. 

We haven't analyzed the issue further this month due to other projects. I have some other ideas on debugging this issue but will look at it in the next week. Hopefully, this should give us some hint about where the issue could be.

Again, thank you for taking the time to solve the issue.

Best regards,
Michal Vatecha.

dji-kyle, 2020-09-27:
Hi，@L4ncelot The last log record you provided is incomplete, and there is no SDK data package, so input data and loop data cannot be compared. And our internal test did not reproduce the user's problem.
Could you please comform if this issue is still exist? 
If the problem still exists, please provide the complete machine LOG of the reproduction steps and corresponding sorties again.
In addition, colleagues who provide reproduction steps need to attach the corresponding A3 firmware version, OSDK version, development platform architecture and system version.

closed 2021-06-24 by dji-kyle