dji-sdk/Onboard-SDK #510: DJI M100 new device error

keebydeane, 2020-03-31:
I am trying to run the sample app (./djiosdk-flightcontrol-sample UserConfig.txt) however the vehicle will not initialise, I have activated the device using the DJI GO app and the keys are correct with my app that I created. 

Any Ideas? The output is as follows;

argv[1] = UserConfig.txt

argv[1] = UserConfig.txt
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L56: Attempting to open device /dev/ttyS0 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L762: Device Serial No. = 041D281159

STATUS/1 @ parseDroneVersionInfo, L764: Hardware = M100

STATUS/1 @ parseDroneVersionInfo, L765: Firmware = 3.1.10.0

STATUS/1 @ parseDroneVersionInfo, L768: Version CRC = 0xA6453AAC

STATUS/1 @ initSubscriber, L813: Telemetry subscription mechanism is not supported on this platform!

STATUS/1 @ initMFIO, L1034: MFIO is not supported on this platform!

STATUS/1 @ initHardSync, L1162: Hardware Sync is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x3010A00

STATUS/1 @ getErrorCodeMessage, L752: activate
STATUS/1 @ getCMDSetActivationMSG, L801: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1424: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1431: Solutions for NEW_DEVICE_ERROR:
        * Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
        * If this is a new device, you need to activate it through the App or DJI Assistant 2 with Internet
        For different aircraft, the App and the version of DJI Assistant 2 might be different
        For A3, N3, M600/Pro and M100, please use DJI GO App
        For M210 V1, please use DJI GO 4 App or DJI Pilot App
        For M210 V2, please use DJI Pilot App
        For DJI Assistant 2, it's available on the 'Download' tab of the product page
        * If this device is previously activated with another app_id and app_key, you will need to re-activate it again.
        * A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.

STATUS/1 @ getErrorCodeMessage, L752: initVehicle
STATUS/1 @ getCMDSetActivationMSG, L801: NEW_DEVICE_ERROR
Vehicle not initialized, exiting.

keebydeane, 2020-03-31:
The issue was solved by restarting the drone and application, including logging out and logging back in. I was then prompted with activate on the screen of the dji go app

closed 2020-03-31 by keebydeane

dji-dev, 2020-04-01:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/33238'>Zendesk ticket #33238</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Glad that your problem has been solved, and thanks for the solution.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team