dji-sdk/Onboard-SDK #511: Support for Z30 advanced sensing on M210 V1

guischmitd, 2020-04-14:
Is there support for the Z30 on M210 V1 planned? My application would benefit a lot from being able to access frames from the Z30 in realtime on the onboard computer we're embedding.

dji-dev, 2020-04-15:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/33580'>Zendesk ticket #33580</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We usually solve SDK-related problems. According to the information you provide, please send your problem information to support@dji.com. We will arrange for someone to contact you as soon as possible.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

guischmitd, 2020-05-08:
Just to leave a public response for OSDK developers facing this issue, there is **no intention** to add camera stream support for the Z30 on the M210 V1 platform since it's discontinued. Bummer. :/

closed 2020-05-08 by guischmitd