dji-sdk/Onboard-SDK #512: [BUG] Connection issues with Matrice 210RTK and Rapsberry Pi 4

SerjTech, 2020-05-04:
Drone Model: Matrice 210 RTK (v1)
Drone Firmware: V01.02.0450
OSDK version: 3.9
OSDK platform: Raspberry Pi 4

Hi, 

Problem 1: 
I'm having problems connecting a Matrice 210RTK with a Raspberry 4 using OSDK. This is what happens during the connection process: 
1 - Both systems start and the OSDK application starts connecting with the drone. 
2 - Connection fails, even when the OSDK app retrieves the correct drone model and drone firmware version from the drone.
3 - The OSDK app tries reconnecting in a loop until it's finally connected. This retry loop usually takes about 1-2 minutes, but sometimes it goes up to 5 minutes. This results in about 20-40 tries to actually connect.
4 - When the system connects, the app starts working as usual, proving that the app itself works, so it's not a drone or app configuration problem.

Problem 2:
While the app is working, there is a problem with the communication between the drone and the app since most of the data packages the app sends through the OSDK are lost. These lost packages causes the system to be useless when trying to retrieve data in real time, since instead of ~5 samples per second It can only send ~1 per 10 seconds. The system DOES NOT throw a bandwith usage exception so that's not the actual problem. 

Additional data: 
Both problems appear only when working with the Matrice 210 RTK. This system is currently being used with a Matrice 600 PRO and with the 600 it connects in about 10 seconds and is also able to retrieve the real time data at a rate of ~5 samples per second.
When testing, both Matrice 600 and Matrice 210 are using the same SDK configuration.

dji-dev, 2020-05-05:
Agent comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/34181'>Zendesk ticket #34181</a>:<br><br>Our DJI Developer Support agents will be on holiday for the international worker's day (5-1 ~ 5-5), which may cause some delays in responding to your issue. Thank you for you understanding and we will answer you as quickly as possible.

Join DJI's Developer Community in our new Forum for Technical Discussions, Announcements and More:
https://forum.dji.com/forum-139-1.html?from=developer

dji-dev, 2020-05-06:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/34181'>Zendesk ticket #34181</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

it is recommended that you use the 1.2.0301 firmware version, as there may be some compatibility issues with the 450 firmware at present. For compatible firmware versions, please refer to the following chart:

 [image]

Problem 1
==> Normally, this problem and phenomenon does not occur when OSDK is used on M210.
In order to better understand and help you solve the problem, need to know more information such as:
1. System type and version used by Raspberry Pi 4?
2. Hardware connection between Raspberry Pi 4 and M210. USB-TTL (What type of chip)? Or directly using GPIO's UART? It is recommended to provide a picture of the hardware connection.
3. The full log that OSDK could not connect, which could also be picture/video.
4. What is the baud rate used?

Problem 2
==> May I ask if you are using the native sample of DJI OSDK for the test? Such as Telemetry sample?
It is recommended that you provide the full log of the OSDK program running.

In addition, ​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

SerjTech, 2020-05-12:
First of all, thanks for answering ;)

The drone has now the version 1.2.0301 but it doesn't solve the issue. All these logs are captured using this version
Here is the data:

RaspberryPi mdel 4B V1.1. OS Raspbian GNU/ Linux 10 (buster)
It's connected using directly the GPIO's UART. I can't show the hardware connection since the access is not easy. 
The baud rate used is 230400. I've also tried with all the other baud rates showed in DJI assistant SDK section, but this one offered the best results and is the value used for the logs.

Attached there are some logs from the app. All these logs appeared in 3 different connecting attempts in under 30 seconds (when the app is not capable of connecting, it restarts itself and tries again, so the logs are 3 consecutive attempts) There are no changes in configuration, connections or code between these logs.
[image]
[image]
[image]

As can be seen on the logs: At the first try, the program is capable of obtaining drone data, but can't connect further. On the second one, it can't obtain drone data, working worse than the previous attempt. Last, on the third one, it just connects and everything works perfectly.

Usually, the steps 1 and 2 repeat for an undefined amount of time. Sometimes it's only one time, sometimes the loop takes 5 minutes.

I've seen that there is a new OSDK version, but it also says that it's compatible with a firmware version that is not available yet, that's why I haven't tried it yet.

dji-kyle, 2021-06-24:
Update to the latest firmware version and test with OSDK 3.9.
The following forum community is recommanded for you, if the problem persists, or new questions about the DJI SDK,
https://djisdksupport.zendesk.com/hc/en-us/community/topics

closed 2021-06-24 by dji-kyle