dji-sdk/Onboard-SDK #513: 请问航点到达时能不能自动执行一些动作，例如悬停指定时长、自动转动云台等

addriumruss, 2020-05-07:
设备： m600pro
osdk:   最新3.9
环境:   linux
需求:  执行waypoint mission, 希望每个航点到达时，飞控能够自动做一些特殊动作，例如悬停指定时长、自动转动云台（水平、俯仰、聚焦）、自动旋转一圈等，做完这些动作之后才自动飞往下一个航点。
在当前osdk WayPointInitSettings结构中找不到额外设置属性。
请问当前osdk是否能够实现这些功能？ 如果不能，什么时候能够发布新版支持这些功能？  
谢谢~

dji-dev, 2020-05-07:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/34249'>Zendesk ticket #34249</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

可以通过设置WayPointSettings的 commandList 和 commandParameter 实现，其中 commandList 代表航点动作的列表，支持的功能如下图：
 [image]​
commandParameter 代表航点动作的参数列表，例如需要到达航点之后悬停5秒的代码：
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/1V4ctXbO30ftmDtLghXf7J5y2' target='zd_attach'>inline527980800.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/SPHmZOhyUikyqn2dZFmI0lTX2' target='zd_attach'>inline1136182302.png</a>

maxiao99, 2020-06-15:
设备：m600 pro，固件版本 3.2.41.14
osdk：DJI Onboard SDK 3.9
环境： Ubuntu16.04
需求：执行waypoint mission，如何判断设备到达任务中的某一个航点？通过setWaypointEventCallback(missionEventCallback,0)设置回调，但回调函数missionEventCallback不执行，请问是否还有其他方法？
非常感谢，期待您的回复

snakehaihai, 2020-06-15:
可以通过拿GPS topic feedback 和你的任务目标点来判断。 如果你需要到某点call off mission。 可以在osdk 里取消任务

addriumruss, 2020-06-16:
@maxiao99 类似这样, 我已经测过能收到航点到达事件。

void waypoint_reached_callback_handler(Vehicle* vehicle,
		RecvContainer recvFrame, UserData userData);

vehicle->missionManager->wpMission->setWaypointEventCallback(
				waypoint_reached_callback_handler, ptr);

closed 2020-06-16 by addriumruss

Challovactor, 2020-07-14:
> Agent comment from Bruce Cheung in [Zendesk ticket #34249](https://djisdksupport.zendesk.com/agent/tickets/34249):
> 
> 尊敬的客户，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 可以通过设置WayPointSettings的 commandList 和 commandParameter 实现，其中 commandList 代表航点动作的列表，支持的功能如下图：
> [image]​
> commandParameter 代表航点动作的参数列表，例如需要到达航点之后悬停5秒的代码：
> [image]​
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> [inline527980800.png](https://djisdksupport.zendesk.com/attachments/token/1V4ctXbO30ftmDtLghXf7J5y2)
> [inline1136182302.png](https://djisdksupport.zendesk.com/attachments/token/SPHmZOhyUikyqn2dZFmI0lTX2)

请问commandParameter支持的功能在哪里可以查到？

kimberz-git, 2021-01-17:
> Agent comment from Bruce Cheung in [Zendesk ticket #34249](https://djisdksupport.zendesk.com/agent/tickets/34249):
>
> 尊敬的客户，
>
> 您好！感谢您联系DJI大疆创新。
>
> 可以通过设置WayPointSettings的 commandList 和 commandParameter [...]

您好，请问您上述贴出来的航点动作列表中支持的功能的介绍可以在哪些文档里面查到？