dji-sdk/Onboard-SDK #518: *.cmake can bot be installed

kyleYehh, 2020-05-15:
For osdk4.0, when sudo make install, *.h and *.a can be installed, but it does not install *.cmake automatically. I tried to debug but failed and I think there is something wrong in the CMakeLists.txt.

plangenh, 2020-05-20:
I am also having problems with cmake on a raspberry pi. 
when I try to "`make`" it says `make: *** No targets specified and no makefile found.  Stop.`

dji-dev, 2020-05-29:
@plangenh @kyleYehh  Sorry, according to the information you provided, we are unable to determine the cause of the problem. In order to better understand and help you solve the problem, it is recommended that you click the hyperlink on the right to fill in more detailed problem information (including the necessary log/screenshot/video), https://formcrafts.com/a/dji-developer-feedback-en

closed 2021-06-24 by dji-kyle