dji-sdk/Onboard-SDK #524: ADVANCED_SENSING cannot be disabled on build

jonatanolofsson, 2020-05-30:
Since commit [ee3c237](https://github.com/dji-sdk/Onboard-SDK/commit/ee3c23725ede0199ea139c09bcbdf96a123de087), ADVANCED_SENSING is not a CMake option, but a variable which cannot be disabled.

Further, the advanced_sensing subdirectory seems to be [added regardless of the variable](https://github.com/dji-sdk/Onboard-SDK/blame/f8bbad7889e923edcebe93a9191898a0b0687e8a/osdk-core/CMakeLists.txt#L113).

jonatanolofsson, 2020-05-30:
Further, samples are [built unconditionally](https://github.com/dji-sdk/Onboard-SDK/blob/f8bbad7889e923edcebe93a9191898a0b0687e8a/CMakeLists.txt#L70), making the build [require libusb and ffmpeg](https://github.com/dji-sdk/Onboard-SDK/blob/f8bbad7889e923edcebe93a9191898a0b0687e8a/sample/platform/linux/CMakeLists.txt#L94-L95)

dji-dev, 2020-06-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35239'>Zendesk ticket #35239</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Advanced_sensing  can't disabled to build OSDK with version 4.0,
if necessary ， please go to : https://formcrafts.com/a/dji-developer-feedback-en and fill in relevant information。

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle