dji-sdk/Onboard-SDK #526: Cannot set home location with onboardsdk3.9 & A3

matspetter, 2020-06-01:
We are trying to set the home location using the onboardsdk3.9 together with an A3 (with the latest fw) but it doesn't work at all.

We do it like this:
```
FlightController::HomeLocation hl;
hl.latitude = lat;
hl.longitude = lon;
vehicle->flightController->setHomeLocationSync(hl, 1);
```

But we always get a timeout on the call.

dji-dev, 2020-06-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35299'>Zendesk ticket #35299</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

About your question ,  Set customized home location failed reason may as follows:
1. The distance between new home location and last home location is larger than MAX_FLY_RADIUS(20km).
2. Set init home location failed after start aircraft.

then I suggest run the sample: djiosdk-flightcontrol-sample, and option c, to check "setNewHomeLocation" at the same environment.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

matspetter, 2020-06-02:
Yes I have tried the sample and it fails. Here is the full log:
```
STATUS/1 @ main, L92: Turn off rtk switch failed, ErrorCode is:       3
STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=10, nTopics=2.
Motors spinning...
Ascending...
Successful takeoff!

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=2.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=2.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 34.
STATUS/1 @ startPackage, L355: Package 0 info: freq=1, nTopics=2.

STATUS/1 @ getErrorCodeMessage, L752: startPackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_UID

**STATUS/1 @ getErrorCodeMessage, L752: getHomeLocation
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_UID**

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_PACKAGE_DOES_NOT_EXIST

STATUS/1 @ setGoHomeAltitude, L153: Set go home altitude failed, ErrorCode is:       3
STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=2.
Task timeout!

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=10, nTopics=4.

STATUS/1 @ goHomeAndConfirmLanding, L190: Start go home action
STATUS/1 @ checkActionStarted, L99: DISPLAYMODE: 15 ...
STATUS/1 @ goHomeAndConfirmLanding, L208: Finished go home action
STATUS/1 @ goHomeAndConfirmLanding, L211: Start landing action
:
:
```

as you can see it succeeds to takeoff and do some flying but when it tries to set the new home location it fails. Actually it never tries to set it since getHomeLocation fails.

Illegal UID sounds rather serious to me. AS I said before. I use onboardsdk 3.9.0 and an A3

matspetter, 2020-06-02:
I would also like to add that when I did this it was by using the "DJI Assistant 2.0 Autopilot" and its builtin **simulator** so it is not a real physical flight.

dji-dev, 2020-06-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35299'>Zendesk ticket #35299</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

I have comfirmed that the function "setNewHomeLocation" not support A3.
Sorry for any  inconvenience.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

sharvashish, 2020-08-26:
If A3 does not support this feature, does M210 support it?
What all autopilots support this feature of setting a new home location.

dji-kyle, 2021-06-24:
M210 v2 / M300 support this feature/API to set  new home location.

closed 2021-06-24 by dji-kyle