dji-sdk/Onboard-SDK #528: Unknown deviceStatus in TOPIC_CONTROL_DEVICE

jeffav, 2020-06-06:
In the telemetry struct SDKInfo broadcast on TOPIC_CONTROL_DEVICE, I am receiving a value of 4 for deviceStatus when requesting Onboard control mode.

[This line](https://github.com/dji-sdk/Onboard-SDK/blob/28da15ad1653bbc746888ac1b8ca8a8706d1eaf0/osdk-core/api/inc/dji_telemetry.hpp#L537) only enumerates what a deviceStatus of 0, 1, or 2 means. The same information is found in [the API Documentation](https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1Telemetry_1_1SDKInfo.html#a493c31f8ea34f5bd17808084931b2262). 

In the previous Onboard-SDK (<4.0), I would receive a value of 2 in this situation.

The comment in the code and the documentation are out of date, or there is a bug causing a value of 4 to be returned.

jeffav, 2020-06-06:
OSDK Version: 4.0
Hardware: DJI M300
Companion computer: DJI Manifold 2

dji-dev, 2020-06-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35507'>Zendesk ticket #35507</a>:<br><br>Dear developer,

Thank you for contacting DJI.

I have reproduced the problem. The question needs further analysis and confirms, if need further communication or there is any result, we will contact you immediately.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-kyle, 2021-06-24:
Note: For M300 and M210V2(firmware version V01.00.0690 and later):0->rc  1->app  4->serial

closed 2021-06-24 by dji-kyle