dji-sdk/Onboard-SDK #529: M300 Advanced Sensing cable not recognized as ACM device

jeffav, 2020-06-09:
I am connecting an M300 to a NVidia Xavier NX developer board through the companion computer interface board. The FTDI serial cable connects and I am able to communicate through that. However, the Advanced Sensing cable has an error when connecting. This is the output of `dmesg`:
```
[ 1654.410917] usb 1-2-port4: Cannot enable. Maybe the USB cable is bad?
[ 1655.262959] usb 1-2-port4: Cannot enable. Maybe the USB cable is bad?
[ 1655.263429] usb 1-2-port4: attempt power cycle
[ 1657.642944] usb 1-2-port4: Cannot enable. Maybe the USB cable is bad?
[ 1658.514939] usb 1-2-port4: Cannot enable. Maybe the USB cable is bad?
[ 1658.515355] usb 1-2-port4: unable to enumerate USB device
```

However, I am able to connect the Advanced Sensing cable successfully to other computers such as the DJI Manifold 2 and the NVidia TX2. Here is the output from `dmesg` on the TX2:
```
[   85.922664] tegra-xusb 3530000.xhci: exiting ELPG
[   85.936432] tegra-xusb 3530000.xhci: Firmware timestamp: 2018-12-26 10:30:04 UTC, Version: 55.12 release
[   85.938045] tegra-xusb 3530000.xhci: exiting ELPG done
[   86.164188] usb 1-3: new high-speed USB device number 3 using tegra-xusb
[   86.190139] usb 1-3: New USB device found, idVendor=2ca3, idProduct=001f
[   86.190162] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   86.190178] usb 1-3: Product: e1e
[   86.190193] usb 1-3: Manufacturer: DJI
[   86.190206] usb 1-3: SerialNumber: 123456789ABCDEF
[   86.197419] usb-storage 1-3:1.2: USB Mass Storage device detected
[   86.201795] scsi host2: usb-storage 1-3:1.2
[   86.262620] rndis_host 1-3:1.0 usb1: register 'rndis_host' at usb-3530000.xhci-3, RNDIS device, da:9a:b6:06:c1:e8
[   86.263545] usbcore: registered new interface driver rndis_host
[   86.278727] cdc_acm 1-3:1.4: ttyACM0: USB ACM device
[   86.286005] usbcore: registered new interface driver cdc_acm
[   86.286013] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   86.318472] IPv6: ADDRCONF(NETDEV_UP): usb1: link is not ready
[   87.213548] scsi 2:0:0:0: Direct-Access     Linux    File-Stor Gadget 0409 PQ: 0 ANSI: 2
[   87.225300] scsi 2:0:0:1: Direct-Access     Linux    File-Stor Gadget 0409 PQ: 0 ANSI: 2
[   87.233810] sd 2:0:0:0: [sda] Attached SCSI removable disk
[   87.238875] sd 2:0:0:1: [sdb] Attached SCSI removable disk
```

I have attempted to manually load the cdc_acm and rndis_host modules, but there is no change in the output. How can I connect the Advanced Sensing cable to the NVidia Xavier NX?

dji-dev, 2020-06-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35567'>Zendesk ticket #35567</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry that I want to comfirm the question, is it successed to connect M300 to DJI Manifold 2 using Advanced Sensing cable, but failed to connect M300 to NVidia Xavier NX ?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

jeffav, 2020-06-09:
Yes, the Advanced Sensing cable successfully connected to the DJI Manifold 2 and failed on the NVidia Xavier NX.

dji-dev, 2020-06-10:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35567'>Zendesk ticket #35567</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, this problem is beyond the scope of our consultation, we generally solve the functional problems of OSDK itself,
but it is recommended that check connections on the NVidia Xavier NX.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2020-06-11 by jeffav

avashist003, 2020-07-06:
@jeffav Hey, I am trying to get the usb-ttl connection with the M300 drone and the jetson tx2. But I am not able to establish the working connection, I get an error drone activation failed when I launch the dji_flighcontrol script with the userconfig.txt.

I see that you are able to get the serial connection with usb-ttl running, can you share some pointers to what I might be missing.

Running the latest osdk 4.0 and attached screenshot of the error
Enabled the osdk api in the app
Not using the acm port as of now only using the ttyusb0 for the usb-ttl connection.
baud rate are set same in user config and dji assistant app= 921600

[image]

Best,