dji-sdk/Onboard-SDK #537: RTK informations

ADlead, 2020-06-12:
Hello, I have a question. During my developing, can I get the RTK information from M300 RTK or M210 RTK with the help of onboard sdk?

dji-dev, 2020-06-12:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35736'>Zendesk ticket #35736</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

You could get the RTK information from M300 RTK or M210 RTK with OSDK subscribe TOPIC.
reference to website: https://developer.dji.com/onboard-api-reference/group__telem.html

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

snakehaihai, 2020-06-15:
yes you can. for RTK osdk 3.9 and above should have no issue in getting those topic. 3.8 there was osdk bug which you cant obtain RTK topic

closed 2021-06-24 by dji-kyle