dji-sdk/Onboard-SDK #541: RTK base station location

adjablonski, 2020-06-15:
Hi,

I'm using the D-RTK with a M600Pro. Is there any way to get the location of the RTK base station using the OSDK? I've looked through all the subscription topics and documentation and I'm not seeing anything. You can access the location of the base with the DJI Go App, but it would be great if there was an option to get it from the OSDK.

dji-dev, 2020-06-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35832'>Zendesk ticket #35832</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry that the Onboard-SDK does not support to get the location of the RTK base station at present.
If you have requirements for relevant functionality, It is recommended that you click on the links on the right side of the fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

snakehaihai, 2020-06-17:
Yes you can

Use code an MSDK app. Call and get  RTKBaseStationInformation.

Here you have the RTK information, now you need to pass to OSDK

in osdk there is sth called https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-mobile-communication.html

Use the correct ios or android API to send command to osdk  https://developer.dji.com/onboard-sdk/documentation/appendix/mobile-onboard-APIs.html

Hope this answer your question. 

Regards
Dr Yuan Shenghai

closed 2021-06-24 by dji-kyle