dji-sdk/Onboard-SDK #543: Jetson TX2 cmake failed: pthread_create not found

avashist003, 2020-06-23:
Hey,
I am building the OSDK on jetson TX2 platform (Ubuntu 18.04, cmake version 3.10.2, gcc version 7.5). I receive error during the cmake process and the terminal shows **pthread_create - not found and gives undefined reference error**.

When I try it on a separate machine (ubuntu 18.04 and same cmake and gcc version) that works. But not on the Jetson platform. Can you advise on what I should be doing to make it work on the jetson ?

avashist003, 2020-06-23:
had to manually install some of the ffmpeg libraries and then it compiled

avashist003, 2020-06-23:
had to manually install some of the ffmpeg libraries and then it worked

closed 2020-06-23 by avashist003

nZac, 2020-11-02:
For those coming later, I found this to work: `cmake -DFFMPEG_INCLUDE_DIR:STRING=/usr/lib/aarch64-linux-gnu ../`

Ubuntu 18.04