dji-sdk/Onboard-SDK #545: armv7 linker

benbarkay, 2020-06-28:
Hi there,

I am trying to compile OSDK 4 for armhf however I am unable to do so because I am missing the linker binary for it. It appears that your cmake build expects it at `osdk-core/linker/armv7`, however that directory is not included in the repository.

Would it be possible to obtain an armhf version of the linker binary and include files?

Thank you,
Ben

dji-dev, 2020-06-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36299'>Zendesk ticket #36299</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that the Onboard-SDK4.0 provides linker binary with platform STM32-M4/ARMv8/X86 at present.
If you have requirements for armv7, It is recommended that you click on the links on the right side of the fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-kyle, 2021-06-24:
Update to the latest

closed 2021-06-24 by dji-kyle