dji-sdk/Onboard-SDK #546: Does M210RTK V2 support

wuxianshen, 2020-06-29:
We have developed a onboard software based on OSDK 3.7 and M210 RTK V1.
Then we purchased a M210 RTK V2, but our old code cannot work normally, which prompts:

**"17_44_0 : STATUS/1 @ init, L65: ...Serial started successfully.

17_44_0 : STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 1DADG4J001011V

17_44_0 : STATUS/1 @ parseDroneVersionInfo, L708: Hardware = PM420

17_44_0 : STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.4.3.42

17_44_2 : ERRORLOG/1 @ initVirtualRC, L1085: Virtual RC is not supported on this platform!

17_44_2 : ERRORLOG/1 @ AdvancedSensing, L57: Please make sure the connected drone is a M210 and firmware version is supported.

17_44_2 : STATUS/1 @ activate, L1295: version 0x304032A
Segmentation fault (core dumped)"**

Is this caused by the wrong OSDK version we used? For M210 RTK V2, which version should we use?

dji-dev, 2020-06-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36347'>Zendesk ticket #36347</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For M210 RTK v2, recommand OSDK4.0, firmware: v01.00.0690.
Suggest that you check and update the firmware by DJI asssitant2 and update OSDK to 4.0. and The SDK changed a lot between 3.x and 4.x，you can evaluate the functionality that has been implemented with old version

Thank you for your understanding and support, hope you have a nice day.

Best Regards,

DJI SDK Support Team

wuxianshen, 2020-06-30:
Thank you for your reply! I will do these updates!

closed 2020-06-30 by wuxianshen