dji-sdk/Onboard-SDK #548: Obtaining Altitude/Height data on M100

XisHungry, 2020-06-30:
Hi, currently I am trying to obtain my drone's Longitude/Latitude and Altitude data. I did manage to obtain both Longitude and Latitude and it appears to be decently accurate. However, my altitude data seems to vary with respect to time. I tested it on multiple occasions, during the morning I obtain an altitude of 31 m and in the afternoon it shows 51 m. 

I did not face any issues with how the drone flies vertically as it is quite accurate when I set the coordinates for throttle. Its just that the initial altitude data is always varying.

I was wondering if there is a way to obtain accurate altitude data on the M100? Currently I am using DJI OSDK 3.7 and OSDK ROS 3.7 for outdoor flights.

Thank you for the assistance

dji-dev, 2020-07-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36382'>Zendesk ticket #36382</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For the question of varying altitude, is it occured stable repetition?
If so, It is recommended that provide more relevant information:
1、Firmware version of M100
2、which API(or TOPIC) you used to obtain altitude data ? and is GPS altitude ?

Suggest that update to OSDK3.8 + ROS3.8 and verifying if it will be occurred similarly.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

XisHungry, 2020-07-01:
Hi, Thank you for the reply!
I tested it on 3 different days, and it seems like a stable repetition.

My DJI M100 firmware is V1.3.1.82
I believe i am using (sensor_msgs::NavSatFix) from demo_flight_control.cpp as shown below.

void
localOffsetFromGpsOffset(geometry_msgs::Vector3&  deltaNed,
                         sensor_msgs::NavSatFix& target,
                         sensor_msgs::NavSatFix& origin)
{
  double deltaLon = target.longitude - origin.longitude;
  double deltaLat = target.latitude - origin.latitude;

  deltaNed.y = deltaLat * deg2rad * C_EARTH;
  deltaNed.x = deltaLon * deg2rad * C_EARTH * cos(deg2rad*target.latitude);
  deltaNed.z = target.altitude - origin.altitude;

  ROS_INFO("Lat=%f, Long=%f, Alt=%f", target.latitude, target.longitude, target.altitude);
}

geometry_msgs::Vector3 toEulerAngle(geometry_msgs::Quaternion quat)
{
  geometry_msgs::Vector3 ans;

  tf::Matrix3x3 R_FLU2ENU(tf::Quaternion(quat.x, quat.y, quat.z, quat.w));
  R_FLU2ENU.getRPY(ans.x, ans.y, ans.z);
  return ans;
}

I believe it is using GPS altitude and i know it has a error range of 10 to 20m. Sorry if i did not provide the relevant information as I am still quite new to how ROS works. I would give OSDK 3.8 and ROS 3.8 a try and see if this error still persist. Thank you so much for your time!

dji-dev, 2020-07-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36382'>Zendesk ticket #36382</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Thanks for your detail, For the question,  suggest that update version of OSDK and ROS to verify it, first.
and if it is occuring continous, it is recommanded to provide operations steps and we will try to reproduce it.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-07-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36382'>Zendesk ticket #36382</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry and there is a extra info about your question: obtaining Altitude , maybe help you to comfirm.

From the sample , you get the altitude from the TOPIC : TOPIC_GPS_FUSED ，TOPIC_ALTITUDE_FUSIONED ， reference to code : dji_sdk_node_publisher.cpp,

void DJISDKNode::publish50HzData(Vehicle* vehicle, RecvContainer recvFrame,DJI::OSDK::UserData userData)
​
And about the TOPIC_GPS_FUSED ，TOPIC_ALTITUDE_FUSIONED, you can visit the developer website ：
https://developer.dji.com/onboard-api-reference/group__telem.html
It maybe explained the  varying altitude obtaining by your drone.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

XisHungry, 2020-07-02:
Hi, I have just installed osdk 3.8 and ROS 3.8. It shows the same data as mentioned above. I will continue to experiment and see if the results changes. Is there really no way for the M100 is above to obtained consistent accurate altitude data (either with respect to mean sea level or ECEF frame)?

Thank you so much for your assistance!

dji-dev, 2020-07-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36382'>Zendesk ticket #36382</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Is there really no way for the M100 is above to obtained consistent accurate altitude data (either with respect to mean sea level or ECEF frame)?
--> GPS has a error range of 10 to 20m. And Noticed for M100, OSDK get telemetry data from the drone by broadcast rather than subscribe.
It maybe help you to confirm the data obtained from the M100.
void DJISDKNode::dataBroadcastCallback()

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2020-08-13 by XisHungry