dji-sdk/Onboard-SDK #555: Cannot find -ldjiosdk-core

ADlead, 2020-07-04:
I downloaded the osdk3.9 on my jetson tx2 and was trying to run the Samples. However, when I try to make and build the sample projects (with CMake), I run into the problem:

`/usr/bin/ld: cannot find -ldjiosdk-core`

The CMake file contains these lines:

`add_executable(${PROJECT_NAME} ${SOURCE_FILES})`
`target_link_libraries(${PROJECT_NAME} djiosdk-core)`

I can't find any lib files like the "*djiosdk-core" in my system
My questions are:

    How do I link the djiosdk-core library?

I am developing on M600 with the JETSON TX2, and Ubuntu 16.04

Thank u so much

ADlead, 2020-07-05:
I hv solved this problem, thank u

YardenLevinger, 2020-07-12:
> I hv solved this problem, thank u

How?

ADlead, 2020-07-14:
> 
> 
> > I hv solved this problem, thank u
> 
> How?

Try cmake the cmakelists.txt at the root dir of this project, which is "Onboard-SDK-3.9/CMakeLists.txt". May fix it.

closed 2021-06-24 by dji-kyle