dji-sdk/Onboard-SDK #557: osdk最新版取视频流, opencv播放, fpv视频很清晰, 但是主视频却是一片花屏，请问怎么解决？

addriumruss, 2020-07-06:
环境： 
osdk最新版, m210飞机, 配Z30主镜头, 以及飞机自带fpv镜头。
osdk运行在Ubuntu系统上, 跑的是这个示例程序: camera_stream_callback_sample, 也跑了这个示例: camera_stream_poll_sample, 
把取到的视频流用opencv进行播放, fpv视频很清晰，但是Z30主视频都是一片花屏，完全显示不出画面。

请问该怎么解决?

dji-dev, 2020-07-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36555'>Zendesk ticket #36555</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

注意到您使用的相机为Z30，请问您使用的Z30固件和飞机M210的固件版本为多少呢？
V01.00.00.72固件Z30相机获取到视频流格式为H264 GDR编码格式，此格式无法直接使用ffmpeg进行解码播放。建议您可以确认一下是否为这个原因。若为解码问题，建议您可根据自己的开发环境选用其他解码器进行适配（比如硬件解码器/JM解码器等等）。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-kyle, 2021-06-24:
最新Z30 固件已经解决OSDK解码问题，可以使用SD卡升级至最新版使用ffmpeg解码播放，加参数：-flags2 showall

closed 2021-06-24 by dji-kyle