dji-sdk/Onboard-SDK #567: unable to compile the sample code

wenbozzz, 2020-07-16:
Hi, 

I'm new to the DJI SDK. I followed the instruction on DJI Developer website and came here try to build the sample code on my raspberry pi 4.
I cloned this repo with head points to 777c30c

The cmake output seems fine
```
pi@raspberrypi:~/Onboard-SDK/build $ cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
Found FFMPEG FFMPEG_INCLUDE_DIR = /usr/local/include
Found FFMPEG FFMPEG_LIBRARIES   = /usr/local/lib/libavcodec.a;/usr/local/lib/libavformat.a;/usr/local/lib/libavutil.a;/usr/local/lib/libswscale.a;/usr/lib/arm-linux-gnueabihf/libz.so
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
/home/pi/Onboard-SDK/osdk-core/advanced-sensing/api/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/platform/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/protocol/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/camera_stream/src/home/pi/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/protocol/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/hal/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/platform/linux/inc
-- Found FFmpeg  installed in the system.
--  - Includes:
--  - Libraries: /usr/local/lib/libavcodec.a;/usr/local/lib/libavformat.a;/usr/local/lib/libavutil.a;/usr/local/lib/libswscale.a;/usr/lib/arm-linux-gnueabihf/libz.so
-----------OSDK_LIB_HEADERS_DIR------------
(/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/api/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/payload/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/flight/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/mop/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/platform/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/logger/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/utility/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/linker/armv7/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/filemgr)
-----------ADVANCED_SENSING_HEADERS_DIR------------
(/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/api/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/platform/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/protocol/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/camera_stream/src/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/ori-osdk-core/protocol/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/ori-osdk-core/hal/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/ori-osdk-core/platform/linux/inc)
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Found FFmpeg  installed in the system.
--  - Includes:
--  - Libraries: /usr/local/lib/libavcodec.a;/usr/local/lib/libavformat.a;/usr/local/lib/libavutil.a;/usr/local/lib/libswscale.a;/usr/lib/arm-linux-gnueabihf/libz.so

stereo-vision-new-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/local/include;/usr/local/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto

stereo-vision-single-thread-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/local/include;/usr/local/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto

stereo-vision-multi-thread-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/local/include;/usr/local/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto

camera-stream-poll-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/local/include;/usr/local/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto

camera-stream-callback-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/local/include;/usr/local/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto

[... 12 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Onboard-SDK/build
```

But when I try to complie, it says unable to find `osdk_typedef.h`?
```
pi@raspberrypi:~/Onboard-SDK/build $ make
Scanning dependencies of target advanced-sensing
[... cut ...]
/home/pi/Onboard-SDK/osdk-core/api/inc/dji_error.hpp:38:10: fatal error: osdk_typedef.h: No such file or directory
 #include "osdk_typedef.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/build.make:63: osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/api/src/dji_advanced_sensing.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:148: osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

I'm not sure how to sovle this, any help would be appreciated. 
Thanks

dji-dev, 2020-07-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36969'>Zendesk ticket #36969</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

As the error shows,  /home/pi/Onboard-SDK/osdk-core/api/inc/dji_error.hpp:38:10: fatal error: osdk_typedef.h: No such file or directory #include "osdk_typedef.h", it seems can't find the linker lib.

It is recommended to confirm and use 64_bit ubuntu on raspberry pi 4.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wenbozzz, 2021-05-12:
Sorry, this was a while ago, I don't remember🤦🏻‍♂️. Maybe I disabled that advance sensing example in cmake config so it doesn't compile that part? Or maybe I added the include path manually so it did find that header file. @JieQ1996

closed 2021-05-18 by wenbozzz