dji-sdk/Onboard-SDK #570: Cannot find FFMPEG Library on Ubuntu 18 WSL

aygunbaltaci, 2020-07-17:
Hello, 

I have been trying to compile the source code with cmake .. but it cannot find the FFMPEG library. 

[image]

I have installed FFMPEG via apt and is located under: 

[image]

I tried to add the paths (except the last one) as prefix to cmake but did not help. Then I went to FindFFMPEG.cmake and added the paths there and this did not help, either. 

Could you guide me with this problem? Thanks! :)

dji-dev, 2020-07-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37037'>Zendesk ticket #37037</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

According to the picture, it shows the FFMPEG execution. Maybe you should comfirm the lib of ffmpeg has been installed , the install method reference to dev-website: https://developer.dji.com/onboard-sdk/documentation/quickstart/development-environment.html

    sudo apt-get install libavcodec-dev libswresample-dev

BTW, Noticed that the developer enviroment is Ubuntu 18 WSL, The OSDK recommand to build and run with Ubuntu1604 temporarily.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

aygunbaltaci, 2020-07-21:
Hello, 

Thanks a lot for the information. Beside the libraries you recommended, we could solve the FFMPEG issue by installing the libraries below:

`sudo apt install libavcodec-dev libavfilter-dev libavformat-dev libavutil-dev`

Afterwards, we had some other missing libraries: 

`sudo apt install python3-opencv libopencv-dev libusb-dev libusb-1.0-0 libusb-1.0-0-dev`

And we had to define this prefix for cmake: 

`cmake .. -DCMAKE_CXX_FLAGS=-lusb-1.0 -DCMAKE_C_FLAGS=-lusb-1.0`

Then cmake could successfully compile the code in Ubun 18 WSL. 

Best Regards
Aygün

closed 2020-07-21 by aygunbaltaci

peza8, 2020-08-06:
Thank you so much! This helped me get it working on The Nvidia Jetson Xavier Nx too 🙏