dji-sdk/Onboard-SDK #572: Compile Error: Makefile:127: recipe for target 'all' failed

Bottle101, 2020-07-20:
When I was trying to compile&install OSDKv4.0, an error occurred as follows.
`[ 83%] Building CXX object sample/platform/linux/advanced-sensing/camera_h264_callback_sample/CMakeFiles/djiosdk-liveview-sample.dir/__/__/common/dji_linux_environment.cpp.o
[ 83%] Building CXX object sample/platform/linux/advanced-sensing/camera_stream_callback_sample/CMakeFiles/camera-stream-callback-sample.dir/__/__/common/dji_linux_helpers.cpp.o
[ 83%] Building CXX object sample/platform/linux/advanced-sensing/camera_h264_callback_sample/CMakeFiles/djiosdk-liveview-sample.dir/__/__/common/dji_linux_helpers.cpp.o
[ 83%] Building CXX object sample/platform/linux/advanced-sensing/camera_stream_poll_sample/CMakeFiles/camera-stream-poll-sample.dir/camera-stream-poll-sample.cpp.o
[ 84%] Building CXX object sample/platform/linux/advanced-sensing/stereo_vision_multi_thread_sample/CMakeFiles/stereo-vision-multi-thread-sample.dir/utility_thread.cpp.o
CMakeFiles/payloads_download_sample.dir/download_sample.cpp.o: In function `fileListReqCB(E_OsdkStat, DJI::OSDK::FilePackage, void*)':
/home/bottle/Apps/Onboard-SDK-master/sample/platform/linux/payloads/download_sample.cpp:40: undefined reference to `DJI::OSDK::printMediaFileMsg(DJI::OSDK::MediaFile)'
CMakeFiles/payloads_download_sample.dir/download_sample.cpp.o: In function `main':
/home/bottle/Apps/Onboard-SDK-master/sample/platform/linux/payloads/download_sample.cpp:117: undefined reference to `DJI::OSDK::getMediaFileName(DJI::OSDK::MediaFile)'
collect2: error: ld returned 1 exit status
sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/build.make:397: recipe for target 'bin/payloads_download_sample' failed
make[2]: *** [bin/payloads_download_sample] Error 1
CMakeFiles/Makefile2:860: recipe for target 'sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/all' failed
make[1]: *** [sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 84%] Building C object sample/platform/linux/advanced-sensing/camera_stream_callback_sample/CMakeFiles/camera-stream-callback-sample.dir/__/__/hal/osdkhal_linux.c.o
[ 84%] Building CXX object sample/platform/linux/advanced-sensing/stereo_vision_multi_thread_sample/CMakeFiles/stereo-vision-multi-thread-sample.dir/image_process_container.cpp.o
[ 85%] Building C object sample/platform/linux/advanced-sensing/camera_stream_callback_sample/CMakeFiles/camera-stream-callback-sample.dir/__/__/osal/osdkosal_linux.c.o
[ 85%] Building CXX object sample/platform/linux/advanced-sensing/camera_stream_callback_sample/CMakeFiles/camera-stream-callback-sample.dir/camera-stream-callback-sample.cpp.o
[ 85%] Building C object sample/platform/linux/advanced-sensing/camera_h264_callback_sample/CMakeFiles/djiosdk-liveview-sample.dir/__/__/hal/osdkhal_linux.c.o
[ 86%] Building C object sample/platform/linux/advanced-sensing/camera_h264_callback_sample/CMakeFiles/djiosdk-liveview-sample.dir/__/__/osal/osdkosal_linux.c.o
[ 86%] Building CXX object sample/platform/linux/advanced-sensing/camera_h264_callback_sample/CMakeFiles/djiosdk-liveview-sample.dir/main.cpp.o
[ 86%] Linking CXX executable ../../../../../bin/djiosdk-liveview-sample
[ 87%] Linking CXX executable ../../../../../bin/stereo-vision-new-sample
[ 87%] Linking CXX executable ../../../../../bin/stereo-vision-single-thread-sample
[ 88%] Linking CXX executable ../../../../../bin/camera-stream-poll-sample
[ 88%] Built target djiosdk-liveview-sample
[ 88%] Built target stereo-vision-new-sample
[ 89%] Linking CXX executable ../../../../../bin/stereo-vision-multi-thread-sample
[ 89%] Linking CXX executable ../../../../../bin/camera-stream-callback-sample
[ 89%] Built target stereo-vision-single-thread-sample
[ 89%] Built target camera-stream-poll-sample
[ 89%] Built target stereo-vision-multi-thread-sample
[ 89%] Built target camera-stream-callback-sample
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

When I run `cmake ..`, it succeed but something in OpenCV cannot be found. I do not know if it would help to solve the problem, so I put them here anyway.
`stereo-vision-depth-perception-sample...
-- Found OpenCV 3.3.1 installed in the system, will use it for depth perception sample
--  - Includes: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_viz;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_cvv;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_hdf;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
-- Did not find ximgproc in OpenCV
-- Did not find viz3d in OpenCV
-- Did not find required libraries, stereo vision depth perception sample will not be compiled.
-- Configuring done
-- Generating done
`

lorenzave, 2020-07-20:
Hi! 
Same error here, it seems that DJI::OSDK::printMediaFileMsg(DJI::OSDK::MediaFile) and DJI::OSDK::getMediaFileName(DJI::OSDK::MediaFile) definitions are missing and they are used for the sample. Breaking the whole build. It seems that the last PR from fixing the download files actually broke it :D I wanted to test that functionality so eager to see the fix soon. 

Best regards, I attach the error if needed. 

`[ 76%] Linking CXX executable ../../../../bin/payloads_download_sample
[ 77%] Building C object sample/platform/linux/advanced-sensing/camera_stream_poll_sample/CMakeFiles/camera-stream-poll-sample.dir/__/__/hal/osdkhal_linux.c.o
[ 77%] Building C object sample/platform/linux/advanced-sensing/camera_stream_poll_sample/CMakeFiles/camera-stream-poll-sample.dir/__/__/osal/osdkosal_linux.c.o
[ 77%] Building CXX object sample/platform/linux/advanced-sensing/camera_stream_poll_sample/CMakeFiles/camera-stream-poll-sample.dir/camera-stream-poll-sample.cpp.o
[ 77%] Building CXX object sample/platform/linux/payloads/CMakeFiles/payloads_sync_sample.dir/__/common/dji_linux_helpers.cpp.o
[ 77%] Building CXX object sample/platform/linux/payloads/CMakeFiles/payloads_async_sample.dir/__/__/__/core/src/waypoint_v2_sample.cpp.o
[ 78%] Building C object sample/platform/linux/payloads/CMakeFiles/payloads_sync_sample.dir/__/hal/osdkhal_linux.c.o
[ 79%] Linking CXX executable ../../../../../bin/stereo-vision-new-sample
[ 79%] Building C object sample/platform/linux/payloads/CMakeFiles/payloads_sync_sample.dir/__/osal/osdkosal_linux.c.o
CMakeFiles/payloads_download_sample.dir/download_sample.cpp.o: In function `fileListReqCB(E_OsdkStat, DJI::OSDK::FilePackage, void*)':
/home/alex/dji_sdk/Onboard-SDK/sample/platform/linux/payloads/download_sample.cpp:40: undefined reference to `DJI::OSDK::printMediaFileMsg(DJI::OSDK::MediaFile)'
CMakeFiles/payloads_download_sample.dir/download_sample.cpp.o: In function `main':
/home/alex/dji_sdk/Onboard-SDK/sample/platform/linux/payloads/download_sample.cpp:117: undefined reference to `DJI::OSDK::getMediaFileName(DJI::OSDK::MediaFile)'
collect2: error: ld returned 1 exit status
sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/build.make:265: recipe for target 'bin/payloads_download_sample' failed
make[2]: *** [bin/payloads_download_sample] Error 1
CMakeFiles/Makefile2:951: recipe for target 'sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/all' failed
make[1]: *** [sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 79%] Building CXX object sample/platform/linux/payloads/CMakeFiles/payloads_async_sample.dir/__/common/dji_linux_environment.cpp.o
[ 80%] Building CXX object sample/platform/linux/payloads/CMakeFiles/payloads_async_sample.dir/__/common/dji_linux_helpers.cpp.o
[ 81%] Linking CXX executable ../../../../bin/payloads_sync_sample
[ 81%] Building C object sample/platform/linux/payloads/CMakeFiles/payloads_async_sample.dir/__/hal/osdkhal_linux.c.o
[ 82%] Building C object sample/platform/linux/payloads/CMakeFiles/payloads_async_sample.dir/__/osal/osdkosal_linux.c.o
[ 83%] Building CXX object sample/platform/linux/advanced-sensing/stereo_vision_multi_thread_sample/CMakeFiles/stereo-vision-multi-thread-sample.dir/utility_thread.cpp.o
[ 83%] Building CXX object sample/platform/linux/advanced-sensing/stereo_vision_multi_thread_sample/CMakeFiles/stereo-vision-multi-thread-sample.dir/image_process_container.cpp.o
[ 83%] Linking CXX executable ../../../../bin/payloads_async_sample
[ 83%] Linking CXX executable ../../../../../bin/stereo-vision-single-thread-sample
[ 83%] Built target payloads_sync_sample
[ 83%] Built target stereo-vision-new-sample
[ 83%] Built target payloads_async_sample
[ 83%] Built target stereo-vision-single-thread-sample
[ 84%] Linking CXX executable ../../../../../bin/camera-stream-poll-sample
[ 84%] Built target camera-stream-poll-sample
[ 85%] Linking CXX executable ../../../../../bin/stereo-vision-multi-thread-sample
[ 85%] Built target stereo-vision-multi-thread-sample
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
`

alduxvm, 2020-07-20:
I get a similar error and I narrow it down to the `download-sample`:

```
.
.
.
.
[ 89%] Built target djiosdk-battery-sample
[ 91%] Built target op_upload_sample
[ 93%] Built target op_download_sample
[ 96%] Built target om_download_sample
CMakeFiles/payloads_download_sample.dir/download_sample.cpp.o: In function `fileListReqCB(E_OsdkStat, DJI::OSDK::FilePackage, void*)':
/home/aldux/Onboard-SDK/sample/platform/linux/payloads/download_sample.cpp:40: undefined reference to `DJI::OSDK::printMediaFileMsg(DJI::OSDK::MediaFile)'
CMakeFiles/payloads_download_sample.dir/download_sample.cpp.o: In function `main':
/home/aldux/Onboard-SDK/sample/platform/linux/payloads/download_sample.cpp:117: undefined reference to `DJI::OSDK::getMediaFileName(DJI::OSDK::MediaFile)'
collect2: error: ld returned 1 exit status
sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/build.make:401: recipe for target 'bin/payloads_download_sample' failed
make[2]: *** [bin/payloads_download_sample] Error 1
CMakeFiles/Makefile2:822: recipe for target 'sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/all' failed
make[1]: *** [sample/platform/linux/payloads/CMakeFiles/payloads_download_sample.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

A patch is needed for that sample I presume...

dji-dev, 2020-07-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37103'>Zendesk ticket #37103</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

So sorry for inconvenience to you, we have noticed the problem and will fixed it later.
If it is needed , you can pull the branch4.0 code not the master.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/j24Mt0FH0z8tAfpsQJxF77bBQ' target='zd_attach'>inline-683931262.png</a>

lorenzave, 2020-07-21:
> Agent comment from kyle.cai in [Zendesk ticket #37103](https://djisdksupport.zendesk.com/agent/tickets/37103):
> 
> Dear developer ,
> 
> Thank you for contacting DJI.
> 
> So sorry for inconvenience to you, we have noticed the problem and will fixed it later.
> If it is needed , you can pull the branch4.0 code not the master.
> [image]​
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team
> [inline-683931262.png](https://djisdksupport.zendesk.com/attachments/token/j24Mt0FH0z8tAfpsQJxF77bBQ)

Looking forward to test the download feature!!

lorenzave, 2020-07-21:
Hi, I am able to compile and use most of the samples but still the download doesnt fully work. I can download files from a m300 but only partially. I can only see the top part of the image, like 5-10%.

I just have one single image from the H20T camera, which are 3 in total and this is the output:

`[1901205.038]STATUS/1 @ activate, L1259: Activation successful

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
a
[1901212.387]STATUS/1 @ main, L84: Play back mode setting......
[1901212.387]STATUS/1 @ main, L88: Get liveview right......
[1901212.389]STATUS/1 @ printErrorCodeMsg, L929: Execute successfully.
[1901212.389]STATUS/1 @ main, L92: Try to download file list  .......
[1901212.389]STATUS/1 @ fileListMonitorTask, L137: OSDK download monitor task created.
[1901212.425]STATUS/1 @ OnReceiveDataPack, L753: [First pack] get the first pack
[1901212.425]STATUS/1 @ parseFileList, L493: Unpack datapack seq(0)
[1901212.425]STATUS/1 @ parseFileList, L505: ###data->amount = 3, data->len = 281
[1901212.425]STATUS/1 @ parseFileList, L548: extData->id = 3, 
[1901212.425]STATUS/1 @ parseFileList, L548: extData->id = 3, 
[1901212.425]STATUS/1 @ parseFileList, L548: extData->id = 3, 
[1901212.425]STATUS/1 @ SendAbortPack, L803: SendAbortPack
[1901212.425]STATUS/1 @ fileListReqCB, L34: ##[Download main camera file list] : ret = 0 
[1901212.425]STATUS/1 @ fileListReqCB, L37: file_list.type = 0
[1901212.425]STATUS/1 @ fileListReqCB, L38: file_list.media.size() = 3
[1901212.425]STATUS/1 @ printMediaFileMsg, L195: ##File [DJI_0060.jpg]Index-262160; 898.7KB; 2020-7-21_14-5-13; Photo details : CameraOrientation::DEFAULT; PhotoRatio::RATIO_4COLON3;
[1901212.425]STATUS/1 @ printMediaFileMsg, L195: ##File [DJI_0061.jpg]Index-262161; 6.9MB; 2020-7-21_14-5-14; Photo details : CameraOrientation::DEFAULT; PhotoRatio::RATIO_4COLON3;
[1901212.425]STATUS/1 @ printMediaFileMsg, L195: ##File [DJI_0062.jpg]Index-262162; 5.2MB; 2020-7-21_14-5-14; Photo details : CameraOrientation::DEFAULT; PhotoRatio::RATIO_4COLON3;
[1901212.425]STATUS/1 @ fileListRawDataCB, L700: Finish req filelist task, reset downloadState to be DOWNLOAD_IDLE
[1901214.398]STATUS/1 @ printErrorCodeMsg, L929: Execute successfully.
[1901214.398]STATUS/1 @ main, L142: Sample end ...

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
b
[1901221.161]STATUS/1 @ main, L102: Download file number : 3
[1901221.161]STATUS/1 @ main, L105: playback mode......
[1901221.161]STATUS/1 @ main, L109: Get liveview right......
[1901221.169]STATUS/1 @ printErrorCodeMsg, L929: Execute successfully.
[1901221.169]STATUS/1 @ main, L114: Try to download file list  .......
[1901221.169]STATUS/1 @ main, L120: targetFile.fileIndex = 262160, localPath = ./DJI_0060.jpg
[1901221.169]STATUS/1 @ startReqFileData, L413: currentLogFilePath = ./DJI_0060.jpg
[1901221.169]STATUS/1 @ fileDataMonitorTask, L181: OSDK download filedata monitor task created.
[1901221.169]STATUS/1 @ printFileDownloadStatus, L132: [Complete rate : 0.0%] (--	kB/s	 recv:	0 packs	 loss:	0 packs) 
[1901221.211]STATUS/1 @ OnReceiveDataPack, L753: [First pack] get the first pack
Preparing File : ./DJI_0060.jpg
[1901221.211]STATUS/1 @ init, L21: fd = 13
[1901221.247]STATUS/1 @ OnReceiveDataPack, L754: [Skip packs]------------------->skip seq : lastSeq = 390, rsp->seq = 415
[1901221.247]ERRORLOG/1 @ fileDataRawDataCB, L730: pack loss !!!
[1901221.247]STATUS/1 @ SendAbortPack, L803: SendAbortPack
[1901221.247]ERRORLOG/1 @ fileDataReqCB, L50: ##Download file data failed. 
[1901221.247]STATUS/1 @ fileDataRawDataCB, L742: Finish req filedata task, reset downloadState to be DOWNLOAD_IDLE
[1901221.250]STATUS/1 @ printFileDownloadStatus, L132: [Complete rate : 42.3%] (--	kB/s	 recv:	392 packs	 loss:	24 packs) 
[1901223.171]STATUS/1 @ printErrorCodeMsg, L929: Execute successfully.
[1901223.171]STATUS/1 @ main, L131: Prepare to do next downloading ...
[1901224.171]STATUS/1 @ main, L105: playback mode......
[1901224.171]STATUS/1 @ main, L109: Get liveview right......
[1901224.174]STATUS/1 @ printErrorCodeMsg, L929: Execute successfully.
[1901224.174]STATUS/1 @ main, L114: Try to download file list  .......
[1901224.174]STATUS/1 @ main, L120: targetFile.fileIndex = 262161, localPath = ./DJI_0061.jpg
[1901224.174]STATUS/1 @ startReqFileData, L413: currentLogFilePath = ./DJI_0061.jpg
[1901224.174]STATUS/1 @ fileDataMonitorTask, L181: OSDK download filedata monitor task created.
[1901224.174]STATUS/1 @ printFileDownloadStatus, L132: [Complete rate : 42.3%] (--	kB/s	 recv:	0 packs	 loss:	0 packs) 
[1901224.267]STATUS/1 @ OnReceiveDataPack, L754: [Skip packs]------------------->skip seq : lastSeq = 428, rsp->seq = 570
[1901224.267]ERRORLOG/1 @ fileDataRawDataCB, L730: pack loss !!!
[1901224.267]STATUS/1 @ SendAbortPack, L803: SendAbortPack
[1901224.267]ERRORLOG/1 @ fileDataReqCB, L50: ##Download file data failed. 
[1901224.267]STATUS/1 @ fileDataRawDataCB, L742: Finish req filedata task, reset downloadState to be DOWNLOAD_IDLE
[1901224.267]STATUS/1 @ OnReceiveDataPack, L753: [First pack] get the first pack
[... 12 lines cut ...]
[1901227.238]STATUS/1 @ OnReceiveDataPack, L753: [First pack] get the first pack
Preparing File : ./DJI_0062.jpg
[... cut ...]
I took a look at the dmesg and found out some warnings occurring **while** running the sample, more intensively when trying to do the file transfer:

`[ 2123.863359] xhci-tegra 3530000.xhci: WARN Event TRB for slot 1 ep 8 with no TDs queued?`
`[ 2124.864001] xhci-tegra 3530000.xhci: WARN Event TRB for slot 1 ep 8 with no TDs queued?`
`[ 2125.864488] xhci-tegra 3530000.xhci: WARN Event TRB for slot 1 ep 8 with no TDs queued?`
`[ 2126.865252] xhci-tegra 3530000.xhci: WARN Event TRB for slot 1 ep 8 with no TDs queued?`
`[ 2127.865930] xhci-tegra 3530000.xhci: WARN Event TRB for slot 1 ep 8 with no TDs queued?`

I hope this can help.

dji-dev, 2020-07-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37103'>Zendesk ticket #37103</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About the problem of OSDK 4.0 master download features, it needs further confirms. and it will be updated if there are any results.
sorry for the inconvenience to you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-07-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37103'>Zendesk ticket #37103</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The OSDK master branch has been updated, but part of the master features has not been test completely, It is recommanded to pull branch 4.0 or before.

Thanks for your feedback.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle