dji-sdk/Onboard-SDK #578: M330 Camera & Advanced Sensing Problem

lueninb, 2020-07-22:
hello,

I am having a problem with Advanced Sensing using the M300.  

We are using a TX2 running 16.04, and a uart and usb cable to connect the drone with the computer.  We would like to grab image frames and store them locally on the TX2.

Activation is successful but the following error is displayed.  Also, when I plug in the USB cable for Advanced Sensing, the feed on the RC freezes.

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [15904]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 81a68aee-cc50-11ea-98a7-00044bf25ee9
process[rosout-1]: started with pid [15917]
started core service [/rosout]
process[vehicle_node-2]: started with pid [15925]
EnableAd: 1
[2012809.108]ERRORLOG/1 @ initVehicle, L162: Failed to initialize ACM Linker channel!
[2012809.124]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADGBV00100JJ
[2012809.124]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.42
[2012809.124]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2012809.126]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2012810.125]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[2012810.325]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2012810.325]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2012810.325]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2012810.325]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2012810.326]STATUS/1 @ RegisterOSDInfoCallback, L535: register result of geting　FC ground station status pushing : 1

[2012810.326]ERRORLOG/1 @ initDJIHms, L900: DJI HMS is not supported on this platform!

[2012810.326]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[2012810.326]STATUS/1 @ activate, L1329: version 0x304032A

[2012810.342]STATUS/1 @ activate, L1353: Activation successful

[ INFO] [1595445678.178143633]: VehicleNode Start
[ INFO] [1595445678.178648951]: Topic startup!
[ INFO] [1595445678.364938024]: Services startup!
[ INFO] [1595445678.477891117]: Use data subscription to get telemetry data!
[ INFO] [1595445678.478001070]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[2012811.702]STATUS/1 @ verify, L244: Verify subscription successful.
[2012811.722]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[2012811.722]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[2012811.763]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[2012811.763]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[2012811.784]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[2012811.784]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[2012811.804]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[2012811.844]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[2012811.844]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[2012811.861]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[2012811.862]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

I would like to grab image frames and store them locally on the TX2.  
1) Do I need Advanced Sensing to do this?
2) What is the proper cable setup, UART, USB C etc?
3) What is the process to get Advanced Sensing working and NOT freeze the RC?
3) What are the correct SDK settings to grab images with or without Advanced Sensing?

Any help here would be greatly appreciated.  Thanks in advance.

dji-dev, 2020-07-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37216'>Zendesk ticket #37216</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、As you described, the drone type is M300, but the log promts some error such as :
[2012810.325]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2012810.325]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2012810.325]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
it seems not match,  if necessary, please comfirm the drone type and error logs again.

2、for your questions:

I would like to grab image frames and store them locally on the TX2.
1) Do I need Advanced Sensing to do this?

   --> Yes, The Advanced Sensing of OSDK support to get video frames from the camera.

2) What is the proper cable setup, UART, USB C etc?

  --> The UART and USB cable connectting the drone with TX2(runs OSDK) both need. and the connections , it is suggest to visit developer webiste: https://developer.dji.com/document/1bb1b2dc-9694-4d6b-ad43-fa69c5a10400

3) What is the process to get Advanced Sensing working and NOT freeze the RC?

 -->Please comfirm your drone type first, if it is M210, you can execute M210tool, reference to : https://developer.dji.com/document/f7161768-60d9-4bfa-8f50-e46cdfbdc3ed

    ./M210ConfigTool --usb-port /dev/ttyACM0 --config-file UserConfig.txt --usb-connected-flight on

4) What are the correct SDK settings to grab images with or without Advanced Sensing?

 -->For OSDK to get video stream , the Advanced Sensing is needed. suggest that you can visit the developer webstite links above.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lueninb, 2020-07-23:
hello,

Thank you for your help.  Yes, we are using the M300.  We want to capture images, not video stream.  Sorry for the confusion.  

1.  Is Advanced Sensing needed to capture images and save them on the TX2?

2.  If I do not need Advanced Sensing, can I use only UART to capture images?  Or do I still need USB?

Our goal is to fly and save the DJI_0001.jpeg etc etc images on the TX2 using the M300.  What process in the code do we use to save the images on the TX2?  Can you provide examples.

Thanks again for all your help.  I really appreciate it.

dji-dev, 2020-07-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37216'>Zendesk ticket #37216</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For M300, there two ways to get the images,
1、The Adanced Sensing provides the way to get the RGB Image, and then handle by the application. reference to links,                                                     https://developer.dji.com/document/8801892e-22be-4c16-a621-87b5fbe18211

2、M300 with OSDK4.0 camera manager provides the way(beta) to download *.JPG from the camera and then handle by application . reference to links, https://developer.dji.com/document/5c6708aa-e123-4c85-97b3-9a689dca6388

And the USB connections is needed,  whether using the features Advanced Sencing or camera manager.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle