dji-sdk/Onboard-SDK #581: waypointMissionV2里的注册回调接口，使用后，会出现osdk与飞机断连的情况

vickiegem, 2020-07-24:
waypointMissionV2里，有被注释的回调函数RegisterMissionStateCallback，在提供的samlpe里使用它，会出现osdk与飞机断连的情况。
用的此接口，
typedef E_OsdkStat (*pFunc)(struct _CommandHandle *cmdHandle,
                      const T_CmdInfo *cmdInfo,
                      const uint8_t *cmdData, void *userData);
void WaypointV2MissionOperator::RegisterMissionStateCallback(pFunc updateMissionState) {
 static T_RecvCmdHandle handle = {0};
 static T_RecvCmdItem item = {0};
 handle.protoType = PROTOCOL_V1;
 handle.cmdCount = 1;
 handle.cmdList = &item;
 item.device = OSDK_COMMAND_FC_2_DEVICE_ID;
 item.cmdSet = V1ProtocolCMD::waypointV2::waypointGetStatePushDataV2[0];
 item.cmdId = V1ProtocolCMD::waypointV2::waypointGetStatePushDataV2[1];
 item.mask = MASK_HOST_DEVICE_SET_ID;
 item.host = 0;
 item.device = 0;
 item.pFunc = updateMissionState;
 item.userData = this;
 bool registerRet = vehiclePtr->linker->registerCmdHandler(&handle);
 DSTATUS("register result of geting mission state pushing : %d\n",
         registerRet);
}
使用后，过10s左右，就会出现断连情况，不用回调函数接口，不会出现此问题。
关于waypointMissionV2,提供的getCurrentState()，获取不到正确的状态，所以才考虑用回调。有什么办法可以正确的获取到路点任务过程中路点任务的状态吗。

dji-dev, 2020-07-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37271'>Zendesk ticket #37271</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

请问您使用的机型和固件版本是多少？另外OSDK您是拉取的哪个分支？

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

vickiegem, 2020-07-24:
机型M300,固件版本V01.00.0109,osdk拉取的22号的代码，这个
[image]

dji-dev, 2020-07-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37271'>Zendesk ticket #37271</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

该问题将进行进一步确认和验证，若有进一步的确认信息或沟通需要将及时与您沟通。另当前master分支版本还在更新中，建议您先使用4.0版本进行验证。抱歉给您相关开发工作带来的不便！

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-kyle, 2021-06-24:
使用最新版固件和最新版OSDK测试

closed 2021-06-24 by dji-kyle