dji-sdk/Onboard-SDK #582: Waypoint mission data not enough

ADlead, 2020-07-24:
During my develpment on osdk with M100, when I tried to upload some waypoints to the drone and run the waypoint mission, I got the error "WAYPOINT_MISSION_DATA_NOT_ENOUGH" and "WAYPOINT_MISSION_POINTS_NOT_ENOUGH" . 

Why does it raise these errors? 

Does it require minimum number of waypoints in waypoint mission? How many waypoints I should upload at least?

My version of osdk is 3.9, and firmware is 3.1.10.0

ADlead, 2020-07-25:
And I also got the error "WAYPOINT_MISSION_INVALID_DATA" at first. But I just uploaded the current gps position of the drone...

Can u help me? Thanke u.

dji-dev, 2020-07-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37282'>Zendesk ticket #37282</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The number of mission waypoints ranges from 2 to 65535. reference the developer website: https://developer.dji.com/document/95217718-ae4b-4ac6-b37b-20fdc738ba5d
 [image]​

You can have a try to upload 2 or more missions.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/TqYBrOauuPArXnmV9qAp2x7Qr' target='zd_attach'>inline765380704.png</a>

closed 2021-06-24 by dji-kyle