dji-sdk/Onboard-SDK #585: 2 ACM Ports, no ttyUSB

aygunbaltaci, 2020-07-29:
Hello,
I'm using Ubuntu 18.04 and I am trying to get the OSDK 3.8.1 test program to run. I follow the tutorial for linux here:https://developer.dji.com/onboard-sdk/documentation/quickstart/run-the-sample.html

I use the Matrice 600 Pro as a drone. From the tutorial I know that I need one ttyUSB and one ttyACM but I don't get a ttyUSB, but I have two ACM ports for this. 
 
[image]
[image]

After I enter the command ./djiosdk-flightcontrol-sample UserConfig.txt I get the error code: 
[2595509.413]ERRORLOG/1 @ initVehicle, L215: Failed to initialize Linker channel
Vehicle not initialized, exiting.

Could you guide me with this problem? Thanks! :)

dji-dev, 2020-07-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37482'>Zendesk ticket #37482</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, There are some questions comfirmed with you.
1、Which cable you used for serial communication?
2、Could you please provide the picture about the hardware connections ?
3、And which platfrom used for the ubuntu1804 ?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

aygunbaltaci, 2020-07-30:
HelloDJI Support Team,
so I use a MicroUSB to USB cable. I plug the MicroUSB into the drone, USB into the PC.
Unfortunately, I didn‘t quite understand your last question. I use Ubuntu 18.04. Here is the exact name: 5.4.0-42-generic 
Here is a picture about the hardware connection. 
[image]

Thank you!

dji-dev, 2020-07-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37482'>Zendesk ticket #37482</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The serial interface for OSDK named as API, you can check the connections as following
 [image]​
The micro USB used to connects with the DJI Assistant2(installed on the pc).

and suggest that you visit the developer website and download the OSDK3.9 document, maybe give you some advice about de hardware connections as above. https://developer.dji.com/document/1be2b992-5862-4eb5-819b-dd6036f76133
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/I6CEnk1AvPUoXcp5DTwIIAj5C' target='zd_attach'>inline725481733.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/YzD7IKdtgC7lAFIZcm9niIQ3k' target='zd_attach'>inline-54013465.png</a>

closed 2021-06-24 by dji-kyle