dji-sdk/Onboard-SDK #586: dji_waypoint_v2.cpp uses the wrong position transformation

matspetter, 2020-07-30:
I have been playing with the master branch of the OnboardSDK to run a waypointv2 mission on an M300 RTK.
However the path flown is not correct. It seems to mixup up the X and Y coordinates somewhere!!

My checkout was on master: hash a0ac2770d082ed8c4751a4eb4de88d09f8b3c95f.

The problem can be fixed in dji_waypoint_v2.cpp and the method missionEncode().

By changing:
    elementEncode<float32_t>(wp.positionX, tempTotalLen, tempPtr);
    elementEncode<float32_t>(wp.positionY, tempTotalLen, tempPtr);

into:
    elementEncode<float32_t>(wp.position**Y**, tempTotalLen, tempPtr);
    elementEncode<float32_t>(wp.position**X**, tempTotalLen, tempPtr);

then everything works fine.

The error can of course also be in the flight controller or somewhere else BUT X and Y are def mixed up somewhere.

If the above solution is considered it shoudl also be done in the same way in missionDecode()

dji-dev, 2020-07-31:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37553'>Zendesk ticket #37553</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The master branch version is still being updated, and we will comfirm this issue later. suggest that you can using the 4.0 branch.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

matspetter, 2020-07-31:
I have tried this on the 4.0 branch as well and the problem is there also. So quite serious.

dji-dev, 2020-08-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37553'>Zendesk ticket #37553</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For this issue , I have verified with OSDK 4.0 branch, but failed to reprodece your problem "mixup up the X and Y coordinates ",
as following , with the sample :djiosdk-waypoint-v2-sample change the mission num to 2.
get the X/Y/Z before the handle as the picture:
    elementEncode<float32_t>(wp.positionX, tempTotalLen, tempPtr);
    elementEncode<float32_t>(wp.positionY, tempTotalLen, tempPtr);
    elementEncode<float32_t>(wp.positionZ, tempTotalLen, tempPtr);
 [image]​
and the simulate result:
 [image]​ [image]
​

May I comfired with you your test result is from the sample: djiosdk-waypoint-v2-sample or the mission uploaded by yourself ?

and suggest that you test using the orign sample , If there are still errors, please provide the operation steps.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/yrZtLPBLqZ77CjlMBlJZvP4Z2' target='zd_attach'>inline-137990090.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/FJK23PqdMHOFriWsdVoZLFJK5' target='zd_attach'>inline-1974359258.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/FRpyyQuubbsvyfOXLKHLZClo8' target='zd_attach'>inline1085166408.png</a>

matspetter, 2020-08-05:
Maybe I should rephrase the problem: It doesn't fly to the correct lat/long coordinates! In the example you ask me to run above there is no way to say how it flies in "real" lat/long.

I have updated the example file waypoint_v2_sample.cpp a bit so that it SHOULD fly : EAST then NORTH then WEST then SOUTH, but it doesn't.

Instead it flies: NORTH then EAST then SOUTH then WEST. With my proposed fix above it flies correctly.

Please try it out and have a look at the RC-controller screen how it flies.

[waypoint_v2_sample.cpp.zip](https://github.com/dji-sdk/Onboard-SDK/files/5028343/waypoint_v2_sample.cpp.zip)

dji-dev, 2020-08-10:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37553'>Zendesk ticket #37553</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the late reply, for the question, the wrong position transformation. we have checked and reproduced it. there maybe some wrong in the API  code,
https://github.com/dji-sdk/Onboard-SDK/blob/master/osdk-core/api/src/dji_waypoint_v2.cpp#L217 and L218

waypointV2Internal.positionX = (waypointV2.longitude - mission[0].longitude) * EARTH_RADIUS *cos(mission[0].latitude);
waypointV2Internal.positionY = (waypointV2.latitude - mission[0].latitude) * EARTH_RADIUS;

we have focued on this issue and submit it to further comfirm, if thers is any result we will contact you soon ,It is recommended that you continue to follow the code updates

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-kyle, 2021-06-24:
Has been fixed

closed 2021-06-24 by dji-kyle