dji-sdk/Onboard-SDK #587: Cannot recognize the NMEA msg received(M210 RTK V2 and Raspberry pi 3)

10857117, 2020-08-02:
Hi,
I'm having a problem after running telemetry-sample code. How can I fix this error messages.
[image]

dji-dev, 2020-08-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37613'>Zendesk ticket #37613</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

From the picture, maybe you use the latest fireware and OSDK 3.8. May I comfirmed with you the firmware version you used is 01.00.0690（check with DJI Assistant2）？
if so, suggest that you update the OSDK version to 4.0, and the NMEA has been updated in vesion 4.0.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

Hepan02, 2021-05-05:
Hi 10857117.

How is your UserConfig.txt looks like? Because I can't recognize DJI on my Pi 4.

closed 2021-06-24 by dji-kyle