dji-sdk/Onboard-SDK #588: Setting DJIWaypointV2FlightPathModeCoordinateTurn and dampingDistance=20 has no effect

matspetter, 2020-08-03:
I'm creating a waypoint mission (with WaypointV2) on an M300 RTK on the simulator with these settings:
waypointType = DJIWaypointV2FlightPathModeCoordinateTurn;
dampingDistance = 20;

The drone should now fly smooth and without stopping but it doesn't. Instead it flies to each waypoint, stops, turns towards the next wp and starts again.

I'm using Onboard SDK 4.0 and flight controller version V01.00.0109

dji-dev, 2020-08-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37638'>Zendesk ticket #37638</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This question needs further comfirmed, if there is any result ,we will contact you soon.
Sorry for the inconvenience to you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

matspetter, 2020-09-01:
Any results on this? I tried the new firmware and osdk v4.0.1 but it still doesn't work.

dji-dev, 2020-09-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37638'>Zendesk ticket #37638</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I'm sorry that this problem is still in the process of scheduling confirmation, and I have submitted the request again to relevant engineers for confirmation. If there is any confirmation result, I will give you feedback in time.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-09-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37638'>Zendesk ticket #37638</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About this issue , suggest that using the lastest firmware version: v01.00.0211 and OSDK 4.0.1 verify again, I have verified and the test results are as follows, I set dampingDistance 200 for the test results to be clear
 [image]
​

And the first waypoint type can't be DJIWaypointV2FlightPathModeCoordinateTurn. if you update the demo code to test ,notice the first waypoint type.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/LdPqPMCWkcuQbUGZIK6tvcbK2' target='zd_attach'>inline-351159546.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/7icjDegmOq48xsjeg3Py2oyLW' target='zd_attach'>inline-1296041935.png</a>

closed 2021-06-24 by dji-kyle