dji-sdk/Onboard-SDK #592: 请问OSDK怎么接收遥控器自定义按钮的事件，谢谢!

addriumruss, 2020-08-06:
环境: M300, OSDK4.0 ubuntu16.04
希望osdk能够接收遥控器自定义按钮的按键事件，不知道具体是哪些接口和类。
请指示，谢谢!

dji-dev, 2020-08-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37772'>Zendesk ticket #37772</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

关于M300 和OSDK4.0获取遥控器数据可参考TOPIC：TOPIC_RC_WITH_FLAG_DATA获取数据，数据结构为：**RCWithFlagData**

该结构可能无法接收遥控器自定义按键事件。若您有相关需求，建议您通过表单填写您的详细需求，我们将根据表单来评估，
链接, https://formcrafts.com/a/dji-developer-feedback-cn

为更好的评估您的需求，建议您提供：
 1.公司背景，
 2.项目描述，
 3.使用到该需求的描述
 4.个人背景介绍

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

addriumruss, 2020-08-07:
不会吧 居然接收不了 
请问填表单是作什么用途，大疆提供定制化飞控和osdk吗？

dji-dev, 2020-08-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37772'>Zendesk ticket #37772</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

提单不是定制化飞控和OSDK，是为了更好的匹配您的具体应用。当前OSDK有提供TOPIC获取遥控器的相关信息，如上封邮件回复，TOPIC_RC_WITH_FLAG_DATA 以及TOPIC_RC_FULL_RAW_DATA，因为不同机型和遥控器物理按键有所差异，您可先查看相关TOPIC及获取数据能否满足您的需求，若不能满足您需求且需要该功能，建议提单描述您的具体需求信息，以供评估是否开放。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

closed 2021-06-24 by dji-kyle