dji-sdk/Onboard-SDK #597: Enable stream on Thermal cameras

IuliuNovac, 2020-08-14:
Hi, 

In the following file, I can see the video stream is done over UDT server - https://github.com/dji-sdk/Onboard-SDK/blob/master/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_link.cpp

#define UDT_SERVER_IP	    "192.168.42.2"
#define UDT_SERVER_PORT_MAIN 	"40001"
#define UDT_SERVER_PORT_FPV  	"40003"

I would like to modify the parameters of the UDT stream, to stream the RGB camera over port 40001, thermal over 40002, zoom over 40003 and remove FPV.

This should be some parameters inside the drone's firmware, so I would like to know if its possible to get access to them.
I am sure this is a quite imporant feature, which for some reason so far hasn't been added?

dji-dev, 2020-08-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38024'>Zendesk ticket #38024</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue is related to the communication protocol, we are sorry that the OSDK communication protocol will no longer be open and not recommand developer to modify it.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

IuliuNovac, 2020-08-17:
> 
> 
> Agent comment from kyle.cai in [Zendesk ticket #38024](https://djisdksupport.zendesk.com/agent/tickets/38024):
> 
> Dear developer ,
> 
> Thank you for contacting DJI.
> 
> This issue is related to the communication protocol, we are sorry that the OSDK communication protocol will no longer be open and not recommand developer to modify it.
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team

Sure, but you have quite a few thermal+RGB cameras. Why did you decide to have a single video stream for both thermal and RGB? Doesn't it make sense to stream them separately?

dji-kyle, 2021-06-24:
Limited by the bandwidth.

closed 2021-06-24 by dji-kyle