dji-sdk/Onboard-SDK #598: M300 OSDK航线只能正常执行一次

ghdmmbyd, 2020-08-15:
您好，
       我用的OnBoardSDK4.0 linux版本（不是ROS），无人机M300最新固件版本，使用V2版本航线功能时第一次能执行，当航线执行完成后，再次同样的流程上传同样的航线航点，总是上传失败，必须得重新启动整个应用程序又可以正常执行一次，使用waypointV2航线例程也是一样的，请尽快帮我分析下这是为什么？谢谢！！

ghdmmbyd, 2020-08-15:
argv[1] = UserConfig.txt
Read App ID
User Configuration read successfully. 

[4045877.202]STATUS/1 @ parseDroneVersionInfo, L1157: Device Serial No. = 1ZNDH6200A2P2X
[4045877.202]STATUS/1 @ parseDroneVersionInfo, L1159: Firmware = 3.4.8.44
[4045877.202]STATUS/1 @ functionalSetUp, L315: Shake hand with drone successfully by getting drone version.
[4045877.203]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[4045878.203]STATUS/1 @ sendHeartbeatToFCTask, L1483: OSDK send heart beat to fc task created.
[4045878.403]STATUS/1 @ FileMgrImpl, L234: register download file callback handler successfully.
[4045878.403]STATUS/1 @ MopPipelineManagerBase, L48: MOP background task now is created .
[4045879.426]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[4045879.426]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[4045879.427]STATUS/1 @ RegisterOSDInfoCallback, L535: register result of geting　FC ground station status pushing : 1

[4045879.427]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[4045879.461]STATUS/1 @ init, L255: Start bulk channel for M300's liveview!
[4045879.461]STATUS/1 @ init, L262: Create task for M300's liveview!
[4045879.465]STATUS/1 @ init, L270: Start bulk channel for M300's perception
[4045879.465]STATUS/1 @ init, L277: Create task for M300's advanced sensing!
[4045879.482]STATUS/1 @ parseDroneVersionInfo, L1157: Device Serial No. = 1ZNDH6200A2P2X
[4045879.482]STATUS/1 @ parseDroneVersionInfo, L1159: Firmware = 3.4.8.44
[4045879.482]STATUS/1 @ AdvancedSensing, L104: Advanced Sensing init for the M300 drone
[4045879.782]STATUS/1 @ init, L283: Start advanced sensing initalization
[4045879.782]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[4045879.782]STATUS/1 @ activate, L1248: osdk policy file updating(1) ......
[4045880.784]STATUS/1 @ activate, L1256: osdk policy file updating(2) ......
[4045880.787]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[4045880.787]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[4045880.789]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[4045880.789]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[4045880.792]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[4045880.792]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 159
[4045880.794]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[4045880.795]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
[4045881.784]STATUS/1 @ activate, L1268: version 0x304082C

[4045881.801]STATUS/1 @ activate, L1292: Activation successful

[4045882.840]STATUS/1 @ verify, L244: Verify subscription successful.
[4045882.860]STATUS/1 @ startPackage, L345: Start package 0 result: 0.
[4045882.860]STATUS/1 @ startPackage, L347: Package 0 info: freq=1, nTopics=1.

[4045882.865]STATUS/1 @ runWaypointV2Mission, L116: Set up subscription successfully!

lat:rad[0.4922961],deg[28.20648639]  lng:rad[1.9702666],deg[112.88796397]
lat:rad[0.4922961],deg[28.20648639]  lng:rad[1.9702677],deg[112.88802514]
lat:rad[0.4922969],deg[28.20653307]  lng:rad[1.9702672],deg[112.88799455]
lat:rad[0.4922969],deg[28.20653307]  lng:rad[1.9702661],deg[112.88793339]
lat:rad[0.4922961],deg[28.20648639]  lng:rad[1.9702656],deg[112.88790281]
lat:rad[0.4922952],deg[28.20643971]  lng:rad[1.9702661],deg[112.88793339]
lat:rad[0.4922952],deg[28.20643971]  lng:rad[1.9702672],deg[112.88799455]

missionInitSettings.missTotalLen = [8]
[4045883.871]STATUS/1 @ initMissionSetting, L309: Init mission setting successfully!

[4045884.876]STATUS/1 @ uploadWaypointMission, L369: Upload waypoint v2 mission successfully!

[4045885.884]STATUS/1 @ downloadWaypointMission, L385: Download waypoint v2 mission successfully!

[4045886.886]STATUS/1 @ uploadWapointActions, L401: Upload waypoint v2 actions successfully!

[4045887.892]STATUS/1 @ startWaypointMission, L416: Start waypoint v2 mission successfully!

[4045907.901]STATUS/1 @ setGlobalCruiseSpeed, L480: Current cruise speed is: 1.500000 m/s

[4045908.906]STATUS/1 @ getGlobalCruiseSpeed, L468: Current cruise speed is: 1.500000 m/s

[4045909.912]STATUS/1 @ pauseWaypointMission, L436: Pause waypoint v2 mission successfully!

[... 30 lines cut ...]
[4046044.001]STATUS/1 @ uploadWapointActions, L401: Upload waypoint v2 actions successfully!

[4046045.004]ERRORLOG/1 @ startWaypointMission, L410: Start waypoint v2 mission ErrorCode:0x110000010036

[4046045.004]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : WaypointV2
[4046045.004]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Unknown
[4046045.004]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : 
[4046065.012]STATUS/1 @ setGlobalCruiseSpeed, L480: Current cruise speed is: 1.500000 m/s

[4046066.017]STATUS/1 @ getGlobalCruiseSpeed, L468: Current cruise speed is: 1.500000 m/s

[4046067.028]ERRORLOG/1 @ pauseWaypointMission, L430: Pause waypoint v2 mission ErrorCode:0x110000010024

[4046067.028]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : WaypointV2
[4046067.028]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : ground station(GS) is not started(used by break/restore) 
[... cut ...]
[4046072.037]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : WaypointV2
[4046072.037]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : ground station(GS) is not started(used by break/restore) 
[4046072.037]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : none
[4046092.052]STATUS/1 @ removePackage, L457: Remove package 0 successful.
[4046092.061]STATUS/1 @ reset, L551: Reset Subscription Successful.

[4046092.061]STATUS/1 @ ~MopPipelineManagerBase, L63: MOP background task now is delete .

dji-dev, 2020-08-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38045'>Zendesk ticket #38045</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

您反馈的这个问题我们已经关注到，且已提交至相关工程师进一步确认。若有任何进展或进一步沟通需要，将及时与您沟通。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

ghdmmbyd, 2020-08-18:
有结果了吗？

dji-dev, 2020-08-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38045'>Zendesk ticket #38045</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

抱歉，该问题可能需要研发工程师进行确认修复，建议您可继续关注GitHub上关于SDK相关修复动态。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

ghdmmbyd, 2020-08-18:
怎么知道是否修复了？这个问题很严重，导致无法正常使用，请相关工程师尽快解决，谢谢

dji-dev, 2020-08-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38045'>Zendesk ticket #38045</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

您可通过关注GitHub上相关动态更新关注问题是否修复，关于waypointV2 stop后再次上传问题已经在master分支上修复更新。您可从master分支获取并验证该问题。stop之后重新init、upload即可。
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/tbjIqMO5814e5wSEwSsIGPnNf' target='zd_attach'>inline1335131657.png</a>

ghdmmbyd, 2020-08-24:
您好，
     感谢！我这边已经解决，请问什么时候出稳定版本？

dji-dev, 2020-08-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38045'>Zendesk ticket #38045</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

OSDK修复该问题的稳定版本4.0.1版本已经在github上发布，您可根据需要下载该版本的SDK。
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/waTndRnCrhUU79hE3xhvxKxVt' target='zd_attach'>inline325151046.png</a>

ghdmmbyd, 2020-08-25:
您好，
    请问下，waypointV2航线上传的航点GPS与实际飞行的位置不对，这个也解决了吗？

dji-dev, 2020-08-26:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38045'>Zendesk ticket #38045</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

请问您所指的航点GPS与实际飞行位置不对是否指该问题：https://github.com/dji-sdk/Onboard-SDK/issues/586
该问题暂未在4.0.1中修复。对于该问题，您可先参照issue 586相关回复修正使用，该问题待进一步确认和验证后再修复至后续稳定版本中。

另关于发布版本中修复的问题，您可参考相关版本release note: https://developer.dji.com/onboard-sdk/downloads/

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

closed 2021-06-24 by dji-kyle