dji-sdk/Onboard-SDK #600: Replace Firmware: DJI N1 Flight Controller DJI inspire for DJI M100 firmware

ccris90, 2020-08-17:
Recently the N1 flight controller of my M100 had a problem. I got another flight controller, but from Inspire. Considering that both are N1, is it possible to replace the inspire firmware with the m100 firmware?

thank you

dji-dev, 2020-08-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38105'>Zendesk ticket #38105</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For the N1 Flight controller and firmware, suggest you contact the aftersales team(support@dji.com), anf they will help you to comfirm the info.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

ccris90, 2020-08-18:
> 
> 
> Agent comment from kyle.cai in [Zendesk ticket #38105](https://djisdksupport.zendesk.com/agent/tickets/38105):
> 
> Dear developer ,
> 
> Thank you for contacting DJI.
> 
> For the N1 Flight controller and firmware, suggest you contact the aftersales team([support@dji.com](mailto:support@dji.com)), anf they will help you to comfirm the info.
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team

Thank you for feedback. I will be make contact with DJI Support.

closed 2021-06-24 by dji-kyle