dji-sdk/Onboard-SDK #608: Not possible to fly higher than 30m nor further away than 30m with the M300/OSDK with the RC switched off

matspetter, 2020-08-19:
We are currently developing a **fully autonomous** system with the new M300 and the onboard sdk (4.0.0).

We **can** takeoff and climb to about 30m with the RC switched off and we **can** fly away about 30m horizontally with the RC switched off but thats it.

We **CANNOT** go any higher nor further away with the RC switched off. If we switch it on we can fly higher and further away.

This is a complete show-stopper for us using the M300 for this use-case (since we do not, and cannot, use the RC in our system).

(all this works fine with the A3 flight-controller)

dji-dev, 2020-08-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38206'>Zendesk ticket #38206</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue is because the design of the drone(such as M600, M210 and M300), without connecting the remote control and APP, the restricted area is within a cylinder with a height of 30m and a radius of 50m. It is to check whether there is a connection between the remote control and APP at the time of taking off.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

ghdmmbyd, 2020-08-24:
> Hi, Maciek here, CTO att [Everdrone](http://www.everdrone.com).
> 
> We’ve built a very competent autonomous UAV system using our own software, the Onboard SDK and partly DJI hardware. We love the DJI hardware and the DJI drones. They’re the best on the market.
> 
> So far we’ve flown over 13.000 autonomous missions using our system. Right now it’s based on the DJI A3 and a DJI drone but we want to move to the M300.
> 
> In order for us to continue using DJI drones we must have to possibility to power on the drone and take off without the RC being connected at all.
> 
> Is there a possibility for you to implement this feature?
> 
> Best,
> Maciek
we are doing this too!!

dji-dev, 2020-08-26:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38206'>Zendesk ticket #38206</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue has been submitted and if there is any result , we will contact you soon.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2020-09-01 by matspetter

maciekd, 2020-09-14:
> Agent comment from kyle.cai in Zendesk ticket #38206 <https://djisdksupport.zendesk.com/agent/tickets/38206>:

Hi, any news on this? 

Best Regards,
  Maciek Drejak