dji-sdk/Onboard-SDK #616: waypoint v2 mission sample not working

XiaoR, 2020-08-24:
M300 newest version, OSDK 4.0 branch commit 94787839d59ea72610bab73e0e7c820e797ac21c

在ARM linux平台简单编译了osdk core和sample后，运行djiosdk-battery-sample， djiosdk-telemetry-sample和djiosdk-flightcontrol-sample都没问题，但是运行djiosdk-waypoint-v2-sample后直接报错ErrorCode:0xFFFFFFFF，Undefined error，没有进一步信息。

./djiosdk-waypoint-v2-sample 
Read App ID
User Configuration read successfully. 

[467038.909]ERRORLOG/1 @ initVehicle, L222: Failed to initialize ACM Linker channel!
[467038.919]STATUS/1 @ parseDroneVersionInfo, L1157: Device Serial No. = XXXXXXXXXXXX
[467038.919]STATUS/1 @ parseDroneVersionInfo, L1159: Firmware = 3.4.8.44
[467038.919]STATUS/1 @ functionalSetUp, L315: Shake hand with drone successfully by getting drone version.
[467038.920]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[467039.920]STATUS/1 @ sendHeartbeatToFCTask, L1483: OSDK send heart beat to fc task created.
[467040.120]STATUS/1 @ FileMgrImpl, L234: register download file callback handler successfully.
[467040.120]STATUS/1 @ MopPipelineManagerBase, L48: MOP background task now is created .
[467041.144]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[467041.144]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[467041.144]STATUS/1 @ RegisterOSDInfoCallback, L535: register result of geting　FC ground station status pushing : 1

[467041.144]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[467041.145]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[467041.145]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[467041.145]STATUS/1 @ activate, L1268: version 0x304082C

[467041.168]STATUS/1 @ activate, L1292: Activation successful

[467042.215]STATUS/1 @ verify, L244: Verify subscription successful.
[467042.235]STATUS/1 @ startPackage, L345: Start package 0 result: 0.
[467042.235]STATUS/1 @ startPackage, L347: Package 0 info: freq=1, nTopics=1.

[467042.240]STATUS/1 @ runWaypointV2Mission, L116: Set up subscription successfully!

[467043.241]ERRORLOG/1 @ initMissionSetting, L214: Init mission setting ErrorCode:0xFFFFFFFF

[467043.241]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[467043.241]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Undefined error
[467043.241]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Unknown error code : 0X%lX, please contact <dev@dji.com> for help.
[467043.268]STATUS/1 @ reset, L551: Reset Subscription Successful.

[467043.269]STATUS/1 @ ~MopPipelineManagerBase, L63: MOP background task now is delete .

dji-dev, 2020-08-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38334'>Zendesk ticket #38334</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

根据您提供的log显示，USB cable未连接或USB通信异常。waypoint v2需要通过USB通信，建议您先检查确认USB连接是否正常。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

XiaoR, 2020-08-25:
谢谢，确实没有连上USB，出错了，已经提了issue #617  请帮忙解答。

closed 2020-08-25 by XiaoR