dji-sdk/Onboard-SDK #621: Not possible to get AirSense/ads-b via OnboardSDK

matspetter, 2020-08-28:
Is it possible to obtain AirSense/ads-b information via OSDK? I cannot find any such possibilities.

dji-dev, 2020-08-31:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38631'>Zendesk ticket #38631</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry the OSDK not support this features to obtain AirSense/ads-b information. If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

matspetter, 2020-09-01:
Ok, have done that.

closed 2020-09-01 by matspetter